prusa3d/Prusa-Firmware-Buddy #2409: [BUG] Installing 4.4.0-beta2 may result in corrupted internal flash [Resolved]

derenma, 2022-08-16:
**Printer type** - [MINI]

**Printer firmware version** - 4.4.0-beta1

**Original or Custom firmware** - Original Beta

**Optional upgrades** - Filament Runout Sensor, BondTech extruder, AITRIP ESP8266 ESP-01S

**USB drive or USB/Octoprint** - USB

**Describe the bug**
After first flash with both bootloader and firmware on USB can result in an "Error #12608: Firmware in the internal flash corrupted! Please reflash the firmware."

**How to reproduce**
Unknown. It just happened.

**Expected behavior**
It flashes both the bootloader and firmware without error

**Crash dump file**
Issue was resolved, so I doubt there is a valid crash dump file

**Resolution**
On the Error screen, do not click RESET or power cycle the machine as neither worked for me and the error kept returning.

To resolve, hold the manual reset button (below main power button) while the error screen is shown. (I held mine down for ~5 seconds for good measure as I do not know its behaviors.)

Prusa-Support, 2022-09-01:
Thank you for reaching us.

This issue duplicates https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2297 and https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2311.

The recommended procedure in case of 'firmware in the internal flash corrupted' error "forcing" the firmware installatiton as described in the sectiton 'Alternative method - older firmware release' of the following guide.
https://help.prusa3d.com/article/firmware-updating-mini-mini_124784

This error would be difficult to debug without reproducing it.
A dump file is generated after a firmware crash and probably overwritten only in case of new crash. I'm not even sure this kind of error would generate a dump file but you can try to export and provide one for our developers to look into it.

Most of time this error could be caused by installing very different firmware versions (non-subsequent firmware releases), but, because we had a few similar reports recently, the investigation is still going on, as you can see in issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2311. Feel free to join the conversatiton.

This issue will be closed as duplicate.

Michele Moramarco
Prusa Research

closed 2022-09-01 by Prusa-Support

krupkaj, 2022-09-04:
I have got the same Error message yesterday. Fortunately I was able to start the flashing again by double click to rotary knob very shortly after reset.

sleeve, 2022-09-05:
I ran into the same problem. Double-clicking right after restarting from the error screen also fixed it for me.