prusa3d/Prusa-Firmware-Buddy #2416: [ENHANCEMENT] Add PA Polyamide/Nylon to filament list

murk-sy, 2022-08-18:
**Printer type** - MINI+

**Is your suggested improvement related to an existing problem? Please describe.**
Printing Nylon on the Mini works (with a lot of asterisks of course), but it is not available in the list of materials. That means you have to pick PC setting to ensure temperatures are high enough for filament changes and for first layer calibration.
Additionally, with filament type check now available, it may cause usability problems further down the road. See also #1919

**Describe the expected functionality**
One of the following options:
1. NYLON becomes available in the filament change menu (it may be recommended to change it to PA? That's more something for the slicer team, I'll make a feature request there too)
2. Add "Custom" options with settable extruder/bed temperatures
3. Add "Unsupported materials" in a submenu; selecting them uses the closest available temperature.

Prusa-Support, 2022-08-27:
Hello! Thank you for the request and for the ideas, sorry for the late response from my side. I brought this to our devs attention to review and possibly implement as soon as possible. In case of more info from our side we'll update on this thread. 

Alessandro Pantaleo
Prusa Research

ThomasCZ, 2022-11-07:
+1 for Nylon

murk-sy, 2023-09-19:
PA has been added with 5.1.0-alpha, closing

Will make a new FR in the future with the material list improvement

closed 2023-09-19 by murk-sy