prusa3d/Prusa-Firmware-Buddy #2421: [FEATURE REQUEST] Allow naming of printer on network

alan412, 2022-08-20:
**Printer type** 
MINI

**Describe the expected functionality**
 I use DHCP for the Prusa mini.   It would be great if I could give it a name such that I could tell prusa slicer to connect to <myprintername>.local and then I wouldn't have to update the IP address when something caused the device to get a different address.

Prusa-Support, 2022-08-30:
Hello! Thank you for your request, I passed this on to our devs for them to further review and possibly implement, in case of more info on this from our side we'll update on this thread. 

Alessandro Pantaleo
Prusa Research

krupkaj, 2022-11-26:
Also need this feature. This is really important when there is more than one printer in the network.

Prusa-Support, 2023-09-16:
This is theoretically already possible but only when flashing the "core" Prusa Link image on [Raspberry Pi Zero W](https://help.prusa3d.com/guide/prusalink-and-prusa-connect-setup-mk3-s-_221744#304787) or [Raspberry Pi 3/4](https://help.prusa3d.com/guide/prusalink-prusa-connect-with-rpi-3-4-usb-mk2-5-s-mk3-s-_469341#470087).

Moreover, the new option to set Prusa Connect as the Host Type in Prusa Slicer will allow this without the use of additional boards as soon as Prusa MINI will be enabled on Prusa Connect (coming soon).
  
[image]

Can this issue be closed?

Michele Moramarco
Prusa Research

alan412, 2023-09-17:
I don't use Prusa Connect as it is on a closed network.   I would rather be able to assign a name and have it work with bonjour (sorry, I forget the generic term for this) so that I could connect to it by name.

alan412, 2024-05-23:
I would still like this feature as I use it on a closed network.

danopernis, 2024-06-27:
Hi @alan412 seems like you are actually requesting mdns support which should be already implemented as indicated in https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3809

alan412, 2024-06-30:
It does work now.   I would recommend it be put into release notes.

closed 2024-07-03 by danopernis

alan412, 2024-07-03:
> ID: <prusa3d/Prusa-Firmware-Buddy/issue/2421/issue_event/13375522116@
> github.com>

Technically this should be closed as done in #3809 instead