prusa3d/Prusa-Firmware-Buddy #2422: [BUG] PrusaSlicer Upload Fails at 100% and corrupted flash drive

Mabritoj, 2022-08-20:
Printer type - MINI

Printer firmware version - 4.4.0-BETA2

Original or Custom firmware - Original

Optional upgrades - Wifi ESP Module (https://www.amazon.com/gp/product/B01EA3UJJ4/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1)

USB drive or USB/Octoprint: USB/WiFi

Describe the bug:
I cannot consistently reproduce this and keep forgetting to gather data every time it happens (screenshots, etc)...but figured since I had it a few times, I will start the report now (I did not find any previous Issue reports...hopefully I am not missing it). I installed a ESP WiFi module and trying out the Prusa Link web interface/PrusaSlicer API Upload, etc with it. On a few occasions now for the following weeks (installed the 4.4.0-BETA2 firmware a few days after it released), PrusaSlicer gets to 100% on the upload and then throws an error about "invalid characters in filename" and then the flash drive is corrupted (using the flash drive that ships with the Mini). I have to reformat the flash drive to make it usable. 

I am uncertain if this happens on ethernet/wired connections

How to reproduce:
I cannot consistently reproduce it and think its random at this point. The last time this happened on, I was slicing the following project: https://www.printables.com/model/58200-nozzle-holder-case-ultra%2Fcomments and the file called Cover.stl. After the issue happened and I reformatted the flash drive...I could not reproduce the issue. At least going by the error message about invalid characters in the filename, the name is "cover.stl" and no special characters, etc.

Prusa-Support, 2022-08-26:
Hello! Thank you for the report. Did this happen on several files, or only on the "cover" file? Did it occur on another USB drive? Which version of Prusa Slicer were you using? I'll bring it to our devs attention, with this info we should be able to narrow this down a bit more. Thank you! 

Alessandro Pantaleo
Prusa Research

Mabritoj, 2022-08-26:
> Hello! Thank you for the report. Did this happen on several files, or only on the "cover" file? Did it occur on another USB drive? Which version of Prusa Slicer were you using? I'll bring it to our [...]

It has happened on several files across different projects (I unfortunately cannot recall which ones or I would link them). I haven't tried other USB drives, just using the silver one the Mini came with. I can certainly format a new one and give it a shot and see if it happens again. 

Using PrusaSlicer 2.4.2

JacobChrist, 2024-05-16:
Not sure if it's related but I uploaded a gcode file with a long file name (BoardMount-OutsideCorner120mm_0.4n_0.2mm_PLA_MK4IS_41m.gcode) to V5.0.0+12061 MK4 printer via the web interface and the Flash drive got corrupted and now looks something like this:
[image]

Prusa-Support, 2024-06-14:
Thanks for bringing this to our attention.

This is potentially related to other forms of data corruption and interferences reported in several ways.
Not to mention that USB flash drives have certain weak points and may fail "by design" for several reasons.

This issue may partially be related to issues like https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3156 or even https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3682, or we may be looking at a whole different problem with the flash drive itself.

This said, in a few cases a full erasing (slow formatting) or other specific flash drive repair software may recover the USB flash drive. However, our developers may appreciate getting a cloned image of the corrupted flash drive via email at reports@prusa3d.com, together with a short description of the problem and of the steps to possibly reproduce the problem.

This issue is based on obsolete beta firmware (stale) and potentially related to other issues, and yet we can't exclude it has little or nothing to do with the firmware. However, it describes data corruption from a different perspective so I would keep it open for now.

Michele Moramarco
Prusa Research

closed 2024-08-22 by github-actions[bot]