prusa3d/Prusa-Firmware-Buddy #2424: [BUG] Print fan not spinning error // Printing fan stops at random

derenma, 2022-08-20:
**Printer type** - MINI

**Printer firmware version** - 4.04b1

**Original or Custom firmware** - Original

**Optional upgrades** - Runout sensor, bondtech extruder, ESP-01s

**USB drive or USB/Octoprint** - USB

**Describe the bug**
During the first few layers, the print fan stops at random. Both hotend fans self-test as OK.

The fan speed error can happen during the first few layers when the fan is supposed to be at 0%. The firmware seems to not like the 0% fan setting now.

**_Workaround_**: I currently have menu timeout disabled and open the "tune" menu after the print starts. After the "fan not spinning error" happens I can press the "continue" button as the UI reverts to the last open menu. (Since I have the tune menu open, the UI reverts to that, not the home screen.) I then make a print fan speed adjustment so I can see the fan turn back on. One the fan is turned back on, I can resume the print.

In some cases, the above steps have to be taken 2-3 times before the print continues as normal.

**How to reproduce**
(In my case) Have the print fan at 0% for the first few layers and the print will halt with a recoverable error.

**Expected behavior**
The fan starts and stops as expected and the fan speed check is done correctly.

**G-code**

[AllanKey_Wall v1.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/9388156/AllanKey_Wall.v1.zip)

murk-sy, 2022-08-24:
Duplicate of #2314

derenma, 2022-08-25:
@murk-sy Closed as dup.

https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2314#issue-1320879737

closed 2022-08-25 by derenma