prusa3d/Prusa-Firmware-Buddy #2425: [BUG] Fan RPM shown on screen is for heatbreak fan, not cooling fan

derenma, 2022-08-21:
**Printer type** - MINI

**Printer firmware version** - 4.4.0b1

**Original or Custom firmware** - Original

**Optional upgrades** - Runout Sensor, Bondtech extruder, ESP-01s

**USB drive or USB/Octoprint** - USB drive

**Describe the bug**
Fan speed shown on screen is for heatbreak fan, not part cooling fan

**How to reproduce**
Run a print

**Expected behavior**
The heatbreak fan speed is kinda useless. Please change to part cooling fan RPM

**Video**

Part fan is stopped; Display still shows ~5k RPM for fan speed.

(Video is part of another issue, but was able to capture this problem as well.)

https://youtu.be/UlY2WSc2rA0?t=83

Spacekoaster, 2022-08-23:
It's not directly related to this topic, but do you have a lot of "printing fan not spinning" error as shown in your video? I never had any and constantly get this error with the 4.4.0-BETA2 firmware.

derenma, 2022-08-24:
@Spacekoaster Yep. Posted an issue here: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2424

Prusa-Support, 2022-09-02:
Thank you for reaching us.

I honestly don't agree that knowing the heatbreak cooling fan RPM is useless.
Theoretically one could print without print cooling fan but not without hotend cooling fan. 🙂
The latter is of crucial importance: the whole print-head block would melt down without it.

This said, footers can be customized 🙂

It doesn't matter if you want to display one fan or the other, both or none.
You can customize the footers as per the section _Footer settings (experimental)_ in the release notes.
https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.0-beta2

I hope you don't mind me closing this issue. In case of further doubts, don't hesitate to ask.

Michele Moramarco
Prusa Research

closed 2022-09-02 by Prusa-Support

derenma, 2022-09-09:
Oh, rock on! I totally missed the Footer settings that I can change now. Thanks.