prusa3d/Prusa-Firmware-Buddy #2428: [BUG] <Prusa Connect aks for API key everytime>

Spacekoaster, 2022-08-22:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version** - [4.4.0-BETA2]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [E3D Revo, Silicone bed mod, Filament Runout Sensor]

**USB drive or USB/Octoprint**
USB drive

**Describe the bug**
Everytime i try to connect to Prusa Connect via my phone (Iphone 11, IOS15.6.1, brower: safari), the Prusa Connect page asks me the API key. I've assigned a static IP to the printer. It happens after i've turned off the printer.

**How to reproduce**
Connect to the local IP adress of the printer

**Expected behavior**
If you've already given the API key with this device, it shouldn't ask you again the API key.

Spacekoaster, 2022-08-22:
Sorry for the confusion, I was talking about "PrusaLink" and not "PrusaConnect". Secondly, the problem happened tonight when the printer was running. I was regularly checking my phone to see the progress, and it suddenly asked me again for the API key.

mix579, 2022-08-26:
Same here. Very irritating

chrishilbert, 2022-08-26:
There's already an enhancement request for this on the prusalink web GitHub.

Spacekoaster, 2022-08-26:
Thanks @chrishilbert, you're totally right!Here's the link of the thread https://github.com/prusa3d/Prusa-Link-Web/issues/222 . I'm closing this issue.

closed 2022-08-26 by Spacekoaster

drakor007, 2022-12-09:
The bug (#2428)is still on version 4.4.0 (final)