prusa3d/Prusa-Firmware-Buddy #2442: [BUG] Change filament is greyed out in the tune menu when connected via USB

napter, 2022-08-26:
**Printer type** - [MINI]

**Printer firmware version** - [e.g. 4.0.5, ...]
4.3.3

**USB drive or USB/Octoprint**
USB/Octoprint

**Describe the bug**
When printing with Octoprint via USB the "change filament" option shows up in the tune menu but is perpetually greyed out. Why is changing/unloading/loading filament not an option? In the MK3S (and every other printer) you can unload and load from the LCD even when connected to a computer. I cannot think of any reason that this would be disabled - perhaps it is a mistake? 

The only solution at the moment is to disconnect and reconnect just to change filament - a pain at any time but will ruin your print if it is in middle of printing - or to return to the computer and issue filament commands from there but that defeats the purpose of the LCD.

**How to reproduce**
Print via USB/octoprint - click menu, tune, see change filament but it is greyed out.

**Expected behavior**
The ability to load/unload/change filament while printing via USB.

Prusa-Support, 2022-09-03:
Thanks for reaching us.

Do the other options work correctly in the tune menu?
Does it do any difference if you downgrade the printer firmware or downgrade/upgrade Octoprint?
Would the option be back available if the print is paused?

Apparently, some of the Prusa-specific firmware features are overridden in the case of a connected "host". Maybe because such fetures are not compatible with the "host" or because the printer can only execute the commands received from the "host".
In latter cases, as speculated in issue https://github.com/prusa3d/Prusa-Firmware/issues/3498 (for instance), often there are ways to execute scripts and reproduce a set of actions from the "host" side.

Michele Moramarco
Prusa Research

napter, 2022-09-04:
Thanks Michele . The other options work. What I am wondering is is why is this option greyed out? The same process works fine on MK3S. 

The issue you linked had to do with sending commands from the host. In this case I am trying to use the LCD. The version of octoprint doesn't matter I have used many versions and it has always been this way on the Prusa Mini. In this case, the firmware is greying out the LCD option and there is no indication as to why. Perhaps someone can look in the code and see what the reason is that the option would be disabled in the LCD?

thijstriemstra, 2022-09-30:
> In this case, the firmware is greying out the LCD option and there is no indication as to why.

I also always found it confusing: because the greyed out option isn't working, I need to perform following steps on prusa mini to change filament:

- go back to the main menu 
- disconnect from octoprint
- go to filament -> change filament
- change and load filament
- return to main menu
- go back to computer with octoprint and connect to prusa mini
- and now I can print again with octoprint

I'd rather not have to disconnect from octoprint at all and use the (greyed out) change filament option on the prusa mini.

Prusa-Support, 2022-10-05:
This all seems to be a problem of compatitbility with Octorpint or caused by the serial communication.
Our developers' efforts are now going in the direction of a stable release for FW 4.4.0, and they are working for the best compatibility with Prusa Link - possibly a valid alternative to Octorpint, directly supported by us.

They will have a look at this at their earliest convenience.
Meantime, it may help to collect more information.

Would the issue persit with older or newer versions of the MINI and Octoprint firmware?
What versions were tested so far?

Michele Moramarco
Prusa Research

thijstriemstra, 2022-10-05:
> Would the issue persit with older or newer versions of the MINI and Octoprint firmware?
What versions were tested so far?

I tested with Octoprint 1.8.0 till 1.8.4 and Prusa MINI+ 4.3.1

napter, 2022-10-05:
So just to clarify again, these are not issues with octoprint. They are
issues with printing over USB using the Prusa Mini. It doesn't matter if
you use octoprint or any other host. The issue is with the Prusa Mini
firmware.

Prusa-Support, 2022-10-13:
There are various concerns about the serial communication, that we are keeping on collecting under the label 'octoprint', probably because it is the most popular alternative way of controlling the printer.

Thank you for confirming the bug with Octoprint (or more generally the serial communication).
Our developers will look into this.

Michele Moramarco
Prusa Research

Prusa-Support, 2022-11-25:
Possibly relates to https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/362.

euggersh, 2023-04-28:
> Thank you for confirming the bug with Octoprint (or more generally the serial communication). Our developers will look into this.

@Prusa-Support, wondering if the developers have looked into this in the 6.5 months since this comment.

Prusa-Support, 2023-05-30:
Unfortunately, Octoprint related issues are more difficult to be addressed now since we no longer offer and maintain the [PrusaPrint Octoprint image](https://help.prusa3d.com/article/prusaprint-rpi-zero-and-octoprint_2180) - also, we are tirelessly working on a possibly better alternative. 🙂 

This said, our developers will possibly into this in the near future, but we don't have a reference timeframe.
Until then, this issue will stay open.

Michele Moramarco
Prusa Research

napter, 2023-05-30:
> Unfortunately, Octoprint related issues are more difficult to be addressed now since we no longer offer and maintain the [PrusaPrint Octoprint image](https://help.prusa3d.com/article/prusaprint-rpi-zero-and-octoprint_2180) - also, we are tirelessly working on a possibly better alternative. slightly_smiling_face

As has been mentioned in this thread this is not an Octoprint issue. This is a usb cable printing issue. The issue will occur when printing over the USB cable - Octoprint happens to be the most common way of doing this but you can test using pronterface, repetier server, or whatever other tool you prefer. 

If you don't support USB printing you can say so but it is not accurate to decline to support these and the other USB cable printing issues by calling them octoprint issues and therefore a third party problem.

Crowlord, 2023-07-06:
> Nearly all Octoprint issues are more generally serial-connection-related.
> More recently we introduced a network/data label for generic issues
> implying data transfer which may be suitable to identify serial connection
> issues too but Octoprint label is more frequently used to identify cases
> involving the use of Octoprint and/or "similar third-party hosts".
> We don't have a specific label for every platform but where Octoprint
> label appears we know that the issue is likely mutual with its
> alternatives: what works for Octoprint most likely works for other serial
> remote control systems too.
>
> The micro USB and serial connection capabilities are complementary
> features, not advertised as a main feature but intended for advanced users.
> In fact, there is no regular or recommended use of it described in the
> Handbook but the following brief definition.
>
> The microUSB port can be used to run third-party solutions, such as
> Octoprint (see https://octoprint.org for more details).
>
> https://cdn.prusa3d.com/downloads/manual/prusa3d_manual_mini_en.pdf
>
> Michele Moramarco
> Prusa Research

Surely being the only printer that DOESN'T fully support Octoprint is a bit
of a downside?

Prusa-Support, 2023-07-26:
The printer is equipped with serial connection capabilities for custom applications but this is not advertised as a main feature of our printers. The firmware is open source though so individuals and the whole community can make use of this characteristic to their advantage.

Unlike Prusa Link & Connect - third-party programs (that are not a product of ours, like Octoprint) are not directly maintained and supported by us but we would gladly possibly look into this in the future.
I hope this clarifies. 

Michele Moramarco
Prusa Research

Crowlord, 2023-07-26:
I believe many of us adopted octoprint when the remote management solution
that was promised never turned up.

This isnt a big firmware ask. We simply wish to be able to load and unload
filament without disconnecting from the Pi.

This is something every other printer does just fine, including your own
Mk2, Mk2.5 and mk3!

Why is this such a big ask?

euggersh, 2023-09-06:
1: It's not an "OctoPrint-related" issue. It's a USB printing-related issue and OctoPrint happens to be the most common way it manifests itself.

2: Why is this issue now closed if it's not fixed yet?

Crowlord, 2023-09-06:
This issue has not been resolved. I am unable to change filament while
connected to usb.

Prusa-Support, 2023-09-14:
closed accidentally
sorry

Prusa-Support, 2024-02-02:
I seem to understand that the problem is solved with the firmware release from 23 Nov 2023 which uniforms the firmware across all Buddy boards (MK4/XL/MINI) and improves the printer compatibility with Octoprint - and supposedly with other terminals connected via USB serial line).

Since [FW 5.1.0](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0) it should be possible for Prusa Firmware and Octoprint to seamlessly work together.

>To make sure everything works correctly, you need to update the Octoprint profile with the following G-codes (in the GCODE Scripts section):
>- After print job is cancelled: M604
>- After print job is paused: M601
>- Before print job is resumed: M602

https://blog.prusa3d.com/prusa-autumn-news-summary-2023-new-firmware-prusaslicer-mk3-5-info-and-more_86935/

Alternative platforms to Octoprint should provide similar parameters to set everything up correctly.

Michele Moramarco
Prusa Research

Prusa-Support, 2024-02-22:
It seems safe to close this issue as solved.
Moreover, the firmware was greatly improved and uniformed across all Buddy boards (MK4/XL/MINI) since 23 Nov 2023 with FW 5.1.0, rendering issues like this potentially obsolete.

Michele Moramarco
Prusa Research