prusa3d/Prusa-Firmware-Buddy #2447: [ENHANCEMENT] Allow overwrite of files on USB stick from PrusaSlicer

mix579, 2022-08-26:
Please move to Prusaslicer or Prusalink, if that's the more appropriate place.

**Printer type** - Mini

**Is your suggested improvement related to an existing problem? Please describe.**
When sending a sliced gcode from Prusaslicer to the Mini via Prusalink, if a file already exists with the same name, the transfer terminates with an error "file already exists". I need to open a webpage, connect to the Mini, and delete the file from storage, before I can re-attempt the upload

**Describe the expected functionality**
When I upload a file from PS via a Prusalink connection to the Mini and it encounters a file with the same file name, it should either just overwrite it, or it could throw up a dialog to confirm to overwrite or cancel.

vorner, 2022-08-29:
Hello

Thank you for the suggestion. Indeed, we know this is uncomfortable. The situation is this way for now because we didn't want to allow overwriting files with the same name by accident and popping up dialogs is a bit more work and needed on all ends (firmware, slicer, PrusaLink, …).

So yes, we want this, it's even being worked on right now. I'm just afraid this won't make it to 4.4.0.

murk-sy, 2022-09-04:
As a workaround (in-firmware, not really as a user), you could just rename the previous files with a file-changed timestamp in a way that hides them. This avoids potentially slow operations such as deletion or having to allocate the same space + some extra for changed file sizes. For most users, I don't think USB storage space is too tight anyways, and I don't think anyone's uploading 300 megabyte prints anytime soon until speeds greatly increase.

Since then a new file is created with the old file name, there will also be no issues with sorting due to modified and created being different. In case of an accidental "overwrite", users can still manually restore the file. The very-lazy-versioning is also a plus.

For example:
`print.gcode` Newest, visible
`print.gcode_20220904121621` Previous, hidden
`print.gcode_20220904121001` Oldest, hidden

I guess "overwrite" wouldn't really be correct - maybe "replace" would be more appropriate for this?

Prusa-Support, 2022-09-23:
This probably concerns both, MINI firmware and Prusa Link, but in the latter repository, we have a similar (or specular) open issue.
https://github.com/prusa3d/Prusa-Link-Web/issues/251

These issues won't necessarily create a conflict between duplicates so we may keep both open for now.

Michele Moramarco
Prusa Research

MightyPork, 2022-11-07:
A trick I'm using for this bug is that I retry and add a digit to the end of the filename (you can edit it right before upload).

The USB drive will keep filling, but with each model being couple MB at most (and larger would take forever to upload over wifi anyway), it's probably fine. Still, if it could do this automatically, that'd be great

GithubUser99999999, 2022-12-13:
+1

This is so annoying. The PrusaSlicer naming scheme would be perfect but unless you change the filename and create even more chaos on your USB stick you cannot upload new gcode. 

Combined with the problem of fixed alphabetic sorting in PrusaLink this creates utter chaos. Even in PrusaLink you cannot delete files. In the end, you always have to remove the thumb drive and manage the files as it was before 4.4.0.

Prusa-Support, 2022-12-22:
This is work in progress.

Meantime, a workaround could be adding _variables_  like `{timestamp}` to the _output file name format_ in Prusa Slicer, so that filenames will be unique.
Alternatively, type a progressive code at the beginning/end of the filename every time while exporting multiple versions or iterations of the gcode, based on your of needs (e.g. _A01, A02... B01... C01..._ or _v010, v020..._).

Michele Moramarco
Prusa Research

closed 2024-06-18 by github-actions[bot]