prusa3d/Prusa-Firmware-Buddy #2449: [BUG] Watchdog Timeout on MK2.5S

wjcampbe, 2022-08-27:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MK2.5S]

**Printer firmware version** - [3.11.0.]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [na]

**USB drive or USB/Octoprint**
  Please let us know if you print via USB flash drive or USB/Octoprint SD card

**Describe the bug**
  I upgraded from 3.11.0 RC1 to full 3.11.0  13a today while troubleshooting a layer shift problem that turned out to be an error in the model.
I printed buddy and it came out perfectly, so I removed the SD to add the model I had repaired and sliced. When I re-inserted the SD I got a single continuous tone  and a screen message "Watchdog Timeout"

**How to reproduce**
  Try removing and reinserting the SD card on a MK2.5S

**Expected behavior**
  Inserting SD card expected sorting message followed by list of files.

**G-code**
na

**Crash dump file**
  Please attach the crash dump file. This will make it easier for us to investigate the bug. Will have to come back with that once I learn how/if I can get a crash dump file off the mk2.5s

**Video**
 na

derenma, 2022-08-28:
@wjcampbe It is possible for you to close your own issue in GitHub.

wjcampbe, 2022-08-28:
you learn something every day - i learned i can close an item and you learned you don't have to offer guidance. Some people can work things out for themselves

closed 2022-08-28 by wjcampbe

derenma, 2022-08-28:
@wjcampbe ¯\_(ツ)_/¯