prusa3d/Prusa-Firmware-Buddy #2459: [BUG] USB drive continually accessed after print finished

DaveMariner, 2022-08-29:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** 
MINI+

**Printer firmware version**
4.4.0 beta-2

**Original or Custom firmware**
Original

**Optional upgrades** 
Joe Prusa signature model 
ESP-01S that flashed but failed to connect to my LAN. 

**USB drive or USB/Octoprint**
USB drive + Network print from PrusaSlicer

**Describe the bug**
When a remote print from PrusaSlicer finishes, the LED in my USB drive continues to flash until I power-cycle the printer. I assume *something* is continually polling the stick and potentially shortening it's lifespan. 

**How to reproduce**
Insert a USB stick with a LED that shows drive access.
Print something from PrusaSlicer over the network (Ethernet cable). 
Wait for the print to finish. 
Invite your friends round for the world's smallest disco. 

**Expected behavior**
I'd expect the LED on the USB stick to stop flashing. 

**G-code**

**Crash dump file**
  Please attach the crash dump file. This will make it easier for us to investigate the bug.

**Video**
  Please attach a video. It usually helps to solve the problem.

Prusa-Support, 2022-09-17:
Thanks for getting in touch about this.

It is possible that Prusa Connect is emitting a "ping" signal until the next operation. This doesn't sound concerning I would say, but our developers could give it a look.

Does the light flickering usually stop after a "normal" print offline?
Does it stop if you exit the "print-finished" screen?

Michele Moramarco
Prusa Research

DaveMariner, 2022-09-19:
Hi Michele

Thanks for the response. Glad to hear that it's probably "only" a ping - I
was mainly concerned that it might be a continuous write of some data that
would be likely to kill USB memory sticks.

The light doesn't stop flickering after an offline print. It also doesn't
stop if you exit the print-finished screen.

It *seems* to be a bit intermittent as to when it gets into this state, but
I think you're correct about Prusa Connect.

The light flashing starts/stops as expected when browsing through the USB
stick to a gcode file, but if you access the printer via a web browser, the
light continues to flash even if you then close the connection to the
printer.

Just speculating, but maybe there's something in the underlying filesystem
side of Prusa Connect that monitors for changes?

Best,

Dave

closed 2024-06-18 by github-actions[bot]