prusa3d/Prusa-Firmware-Buddy #2460: [BUG] Printer has weird behaviours at the end of the print with color changing print

Spacekoaster, 2022-08-29:
**Printer type** - [MINI]

**Printer firmware version** - [4.4.0-BETA2]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [E3D Revo, Silicone bed mod, Filament Runout Sensor, ESP8266]

**USB drive or USB/Octoprint**
USB drive

**Describe the bug**
I've been printing a QR code with 3 different colors (with 2 filament changes during the print). The printer behaves very strangely at the end of the print. At the first try of this print, near the end of the print (between 90 and 95%, I wasn't there when it happened), the head decided to go to back to more of the center of the print, and kept extruding during this. I first thought that it was a "normal" error and didn't think more about it. Second print, the head started printing a full top layer in the middle of the square at the end of the print (of course that wasn't planned again). I thought that the USB drive was maybe the reason of this, so I formated it, sliced my model again and started printing a third try. This time, the print literally stopped in the middle of a layer, and the LCD was telling me that the print was finished (with the usual print again, go back to menu, etc), even tho it was showing 98% and not 100%. I flashed again the printer with stable firmware 4.3.4, printed the same gcode as the third try, and this time the print was perfect and finished as expected.

**How to reproduce**
  See "Describe the bug section"

**Expected behavior**
  The print should normally end without any problem.

**G-code**
I had to change the extension to .txt. Please replace the .txt with .gcode in order to get the exact same gcode as me.
  [ItsLitho_qr-code_0.15mm_PLA_MINI_3h26m.txt](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/9447994/ItsLitho_qr-code_0.15mm_PLA_MINI_3h26m.txt)

**Crash dump file and video**
Unfortunately, I haven't taken any video or saved a crash dump. I'm opened to print again the model if it is necessery for the developpement team.

Prusa-Support, 2022-09-02:
Hello! Thank you for your report, I passed this on to our devs to look into this in more detail and possibly reproduce. Just to make sure, which version of Prusa Slicer, or what program did you use to generate the GCode? Likely not related with that but can be useful info. I checked the GCode and haven't noticed anything irregular there, though. 

Let me know, in case of more info from our side, we'll update on this thread. Thank you! 

Alessandro Pantaleo
Prusa Research

Spacekoaster, 2022-09-02:
Thé gcode has been generated using PrusaSlicer 2.4.2

Prusa-Support, 2022-09-02:
Thank you for the info! 

Alessandro Pantaleo
Prusa Research

closed 2024-06-18 by github-actions[bot]