prusa3d/Prusa-Firmware-Buddy #2489: [FEATURE REQUEST] Start preheat immediately on "Upload and print"

murk-sy, 2022-09-07:
**Printer type** - MINI+
**Optional upgrades** - Filament Runout Sensor

**Is your feature request related to an existing problem? Please describe.**
Based on the latest [blogpost](https://blog.prusa3d.com/dev-diary-what-it-took-to-bring-wi-fi-support-to-original-prusa-mini_70615/), the wireless transfer speed of the print files will never be measured in tens of megabits. It is not unrealistic to expect a 1 minute+ transfer speed for any decently sized print.

**Describe the expected functionality**
1) A print is prepared and sliced
2) The user clicks the "Upload and Print" button
3) Before sending the file, preheat gcode (based on the selected filament) is sent to the printer
4) The file is being uploaded while the printer is already heating up
5) File upload is complete and the print starts normally

### However, a major counterpoint:
This feature may only really be beneficial with short prints (where the heat up time is comparatively long) where the gcode file is either large (which would mean it's not short?) or the transfer speed is very slow.
Additionally, depending on your work process, you may start preheat manually when preparing the bed.

Examples using 0.10 DETAIL preset calculated for 300 kB/s (future expected speed):
- 1 Flexi Rex - 4:43 print - 9870 KB - 33s upload
- 25 50% Flexi Rexis - 1d 4:35 print - 79563 KB - 265s upload

While this will probably make the printer *feel* faster and more optimised, it likely either wouldn't have a major print time improvement, or would heat up the bed for much longer than necessary for longer prints.

**Consider this more a RFQ than a formalized, thought out feature request.**

Prusa-Support, 2022-09-16:
Hello! Thank you for the idea, I passed this on to our devs for them to review and possibly evaluate in more detail. In case of more info from our side on this, we'll update on this thread. 

Alessandro Pantaleo
Prusa Research

rbronco21, 2022-09-27:
I approve of this request... and please include adjustable preheat temps? My PETG likes to ooze and curl back up onto the nozzle. FYI, I like 155C.

ThomasCZ, 2022-10-20:
Time required to preheat and make heat bed calibration can be calculated. It would be nice to not start preparation immediately after the upload starts but in the right moment.

murk-sy, 2023-09-25:
It appears that in [5.1.0-alpha1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v5.1.0-alpha1) using Connect, the print can be started before the file is fully transferred. Not 100% sure if it works the same with just Prusalink, but I think I can already close this issue. From my limited testing it works flawlessly, keep up the good work!

closed 2023-09-25 by murk-sy