prusa3d/Prusa-Firmware-Buddy #2495: [BUG] Stuck head heating

ThomasCZ, 2022-09-09:
I have to start and stop printing to stop head heating after I unload a filament. Expected behavior: Head heating stop immediately after the filament is unloaded.

closed 2022-09-23 by Prusa-Support

ThomasCZ, 2022-09-23:
It is not right, because heat bed is also heating and whole process is waiting for that. When I am unloading a filament, it is just wasting of energy. I think, it is wasting of energy in 100% of cases, because when I select "unload filament", I want to unload a filament. If I wanted to print with another filament, I can select "Change filament". Safety timer does not solve wasting of energy.

Anyway, could you please add at least some "eco" check box, that will not heating heat bed and turn off heating immediately?

Prusa-Support, 2022-09-28:
> It is not right, because heat bed is also heating and whole process is waiting for that. When I am unloading a filament, it is just wasting of energy. I think, it is wasting of energy in 100% of [...]

Duplicate. https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/221#issuecomment-1256177154

Michele Moramarco
Prusa Research