prusa3d/Prusa-Firmware-Buddy #2506: [FEATURE REQUEST] Ability to change the root folder for PrusaLink gcode uploads, and support subfolders for file organization

ShadowDrakken, 2022-09-15:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

If it makes sense, enter what type of printer or upgrade the feature request applies to.
**Printer type** - MINI+
**Optional upgrades** - Filament Runout Sensor, ESP-01S

**Is your feature request related to an existing problem? Please describe.**
Running the 4.4.0-Beta2 firmware for the MINI+, and not thrilled that PrusaLink only supports gcode in the root of the USB drive. This creates quite a mess on the thumb drive.

**Describe the expected functionality**
I keep my thumb drive pretty organized, so it would be nice to be able to change the gcode root, perhaps using the configuration INI file, as well as having the ability to create, delete, and navigate subfolders within that root.

Prusa-Support, 2022-09-16:
Hello! Thank you for your request. This is currently blocked by the fact we use the old Octoprint-compatible API in PrusaLink. We'll resolve this aspect in the future with a new API, but not with future stable FW 4.4 releases. 

Alessandro Pantaleo
Prusa Research

ShadowDrakken, 2022-09-16:
Just as an aside, it appears OctoPrint has supported subfolders since Aug 2020

Prusa-Support, 2022-09-26:
Hello, sorry for the late response. Noted these latest points for our devs, for them to further take a look into - in case of more info from our side, we'll update on this thread, but of course in case of anything more to add, feel free to add more comments. Thank you! 

Alessandro Pantaleo
Prusa Research

MartinMajewski, 2024-04-30:
We have the end of April 2024, and Prusa Connect still ignores subfolders. 

A file with a requested (and already available) subfolder a la "Subfolder/Printfile.bgcode" lands in the root directory, quickly causing the list to become a mess. Reprinting something over the LCD screen is nearly impossible if the file was uploaded a few days earlier.
Moreover, there is no other way to organize the files via a (non-existing) move command in Prusa Connect.

This should be a top priority, IMHO, as it is a productivity stopper for everyone using PrusaConnect for business with a lot of different files over a short time span.

Prusa-Support, 2024-07-02:
Sorry, I believe this issue is obsolete. It is actually solved.

Please give it a try with the directory separator sign `/` as per the pop-up on Prusa Slicer.

[image]
[image]

Michele Moramarco
Prusa Research

closed 2024-07-02 by Prusa-Support