prusa3d/Prusa-Firmware-Buddy #2518: [QUESTION] Where is BETA3 ?

Erik262, 2022-09-22:
Hi Prusa Team 👋
I'm following your progress on the Prusa MINI Firmware very close, and I'm very happy about having nice features with every update.

It's been almost two months since the last BETA update, and I'm asking myself why you don't publish BETA 3? A lot of changes were made since then to proof/check.

Prusa-Support, 2022-09-28:
Thank you for showing your interest, and your comment is appreciated.

We are on it!
The release of this specific firmware took us more than half a year for several reasons, especially the big load of work behind it (you can check the [dev diary](https://blog.prusa3d.com/dev-diary-what-it-took-to-bring-wi-fi-support-to-original-prusa-mini_70615/) on the blog, and the [release notes](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.0-beta2)), so I'm speculating that the next release may take "longer than usual" too.

We don't have an estimate but I have the impression the developers have made big progress 🙂 and the preliminary internal testing has begun.

Thank you for your patience and support.
If your question has been answered in a satisfactory manner, please consider closing this issue.

Michele Moramarco
Prusa Research

Erik262, 2022-09-28:
Thank you for your response. I have already read the "Prusa MINI Odyssey" on the blog. So, in that case, even the BETA would take some more time? (Not wanting to break that title piece on the board to install (maybe) unsigned firmware)

Prusa-Support, 2022-09-29:
Hello 🙂 following up on this, good news - we advanced from Beta and just released the 4.4.0 RC1, full info and resources here: https://github.com/prusa3d/Prusa-Firmware-Buddy/releases 

Alessandro Pantaleo
Prusa Research

closed 2022-09-29 by Erik262