prusa3d/Prusa-Firmware-Buddy #2542: [BUG] <Enter comprehensive title>

czo, 2022-09-29:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - Prusa Mini

**Printer firmware version** - 4.4.0-RC1

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor

**Describe the bug**
The printer seems to me, has exporting an unknown "usb-storage" device, and the kernel try to spin up the disk forever.

kernel log:
Sep 29 19:45:08 dwarf.czo.hu kernel: usb 1-1: new full-speed USB device number 5 using orion-ehci
Sep 29 19:45:08 dwarf.czo.hu kernel: usb 1-1: New USB device found, idVendor=2c99, idProduct=000c, bcdDevice= 1.00
Sep 29 19:45:08 dwarf.czo.hu kernel: usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Sep 29 19:45:08 dwarf.czo.hu kernel: usb 1-1: Product: Original Prusa PRINTER
Sep 29 19:45:08 dwarf.czo.hu kernel: usb 1-1: Manufacturer: Prusa Research (prusa3d.com)
Sep 29 19:45:08 dwarf.czo.hu kernel: usb 1-1: SerialNumber: 00000000001A
Sep 29 19:45:08 dwarf.czo.hu kernel: cdc_acm 1-1:1.0: ttyACM0: USB ACM device
Sep 29 19:45:08 dwarf.czo.hu kernel: usb-storage 1-1:1.2: USB Mass Storage device detected
Sep 29 19:45:08 dwarf.czo.hu kernel: scsi host2: usb-storage 1-1:1.2
Sep 29 19:45:09 dwarf.czo.hu kernel: scsi host2: scsi scan: INQUIRY result too short (5), using 36
Sep 29 19:45:09 dwarf.czo.hu kernel: scsi 2:0:0:0: Direct-Access     Prusa    MINI             1.0  PQ: 0 ANSI: 2
Sep 29 19:45:09 dwarf.czo.hu kernel: sd 2:0:0:0: Attached scsi generic sg1 type 0
Sep 29 19:45:09 dwarf.czo.hu kernel: sd 2:0:0:0: [sdb] Spinning up disk...

root@dwarf:~# cat /proc/scsi/scsi 
Attached devices:
Host: scsi0 Channel: 00 Id: 00 Lun: 00
  Vendor: ATA      Model: Samsung SSD 860  Rev: 4B6Q
  Type:   Direct-Access                    ANSI  SCSI revision: 05
Host: scsi2 Channel: 00 Id: 00 Lun: 00
  Vendor: Prusa    Model: MINI             Rev: 1.0 
  Type:   Direct-Access                    ANSI  SCSI revision: 02

root@dwarf:~# cat /proc/scsi/usb-storage/2 
   Host scsi2: usb-storage
       Vendor: Prusa Research (prusa3d.com)
      Product: Original Prusa PRINTER
Serial Number: 00000000001A
     Protocol: Transparent SCSI
    Transport: Bulk
       Quirks:

lsusb -v:
Bus 001 Device 005: ID 2c99:000c Prusa Original Prusa PRINTER
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x2c99 Prusa
  idProduct          0x000c 
  bcdDevice            1.00
  iManufacturer           1 Prusa Research (prusa3d.com)
  iProduct                2 Original Prusa PRINTER
  iSerial                 3 00000000001A
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0062
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass          2 Communications
      bFunctionSubClass       2 Abstract (modem)
      bFunctionProtocol       0 
      iFunction               0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         2 Communications
      bInterfaceSubClass      2 Abstract (modem)
      bInterfaceProtocol      0 
      iInterface              4 CDC
      CDC Header:
        bcdCDC               1.20
      CDC Call Management:
        bmCapabilities       0x00
        bDataInterface          1
      CDC ACM:
        bmCapabilities       0x02
          line coding and serial state
      CDC Union:
        bMasterInterface        0
        bSlaveInterface         1 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              16
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass        10 CDC Data
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         8 Mass Storage
      bInterfaceSubClass      6 SCSI
      bInterfaceProtocol     80 Bulk-Only
      iInterface              5 MSC
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)

**How to reproduce**
Install 4.4.0-RC1, remove any USB device from the USB-A port and then connect the printer to a computer.

**Expected behavior**
Only the usbserial device will be visible on the computer.

paukstelis, 2022-10-03:
Additionally, I would add that this change also seems to nullify the unique serial number of the USB->UART. This is problematic as a unique serial number is the most useful way of identifying specific printers via udev rules.

meermanr, 2022-10-09:
You can stop the kernel spam by unbinding the SCSI device from its driver:

```shell-session
pi@caboose:~ $ lsusb --tree
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=dwc2/1p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 2: Dev 3, If 0, Class=Communications, Driver=cdc_acm, 12M
        |__ Port 2: Dev 3, If 1, Class=CDC Data, Driver=cdc_acm, 12M
        |__ Port 2: Dev 3, If 2, Class=Mass Storage, Driver=usb-storage, 12M

pi@caboose:/sys/bus/scsi/drivers/sd $ ls
0:0:0:0  bind  uevent  unbind

pi@caboose:/sys/bus/scsi/drivers/sd $ ls -l
total 0
lrwxrwxrwx 1 root root    0 Oct  9 14:49 0:0:0:0 -> ../../../../devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2:1.2/host0/target0:0:0/0:0:0:0
--w------- 1 root root 4096 Oct  9 14:49 bind
--w------- 1 root root 4096 Oct  9 14:33 uevent
--w------- 1 root root 4096 Oct  9 14:49 unbind

pi@caboose:/sys/bus/scsi/drivers/sd $ echo -n '0:0:0:0' | sudo tee  unbind
```

Seemed to "hang" here for ~30s, I think until the kernel gave up on its current attempt to spin up sda, but eventually it returned and the bound device was no longer listed:

```shell-session
pi@caboose:/sys/bus/scsi/drivers/sd $ ls -l
total 0
--w------- 1 root root 4096 Oct  9 14:49 bind
--w------- 1 root root 4096 Oct  9 14:33 uevent
--w------- 1 root root 4096 Oct  9 14:50 unbind

Prusa-Support, 2022-10-18:
Thank you for reporting this.

Our developers have manifested interest in these issues (#2542 and #2558) and are willing to work on a fix in the next few upcoming firmware releases.

Michele Moramarco
Prusa Research

closed 2024-06-18 by github-actions[bot]