prusa3d/Prusa-Firmware-Buddy #2543: On Github the name of the file Buddy_4.4.0-RC1+4164.bbf is wrong - rename-it

catagro, 2022-09-29:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version** - [4.4.0-RC1]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor]

**USB drive**

**Describe the bug**
I try to update the firmware with the new release putting the both files on the flash drive.
The bootloader was updated successfully but after the printer stopped.
The screen message was "**Looking for .bbf file**"
On Github the name of the file **Buddy_4.4.0-RC1+4164.bbf** is wrong.
I have rename-it **Buddy_pre_4.4_4.4.0-RC1+4164.bbf** and the install worked.
Happy printing to all !

**Expected behavior**
  Flashing the firmware

Prusa-Support, 2022-09-30:
Hello! Thank you for your report. So in your case, did you have both files named "Buddy 4.4.0"? I could see the "pre 4.4" was one of the files in the assets on the release page, so you only had the "Buddy 4.4.0" one at the time of the error? Sorry for the confusion on this aspect. Any clarification on this aspect can be useful for us to figure out a resolution for other releases. Thank you! 

Alessandro Pantaleo
Prusa Research

catagro, 2022-09-30:
Dear Mr. Pantaleo
I will try to be more specific describing to you step by step what happened.
I have downloaded both files from github.On github the name of the files are:
 Buddy_4.4.0-RC1+4164 .bbf        andBuddy_update_pre_4.4_4.4.0-RC1+4164.bbf

I have copied the both files to an empty USB stick and I have put the stick on the printer.The printer have recognized the new boot firmware and it have started to update the boot.Updating the boot sequence was fast and ok, unfortunately after that, the printer have not passed to the next step to update the firmware.On the LCD screen the message was "Searching for the bbf file" (if I remember well) 
and nothing happened after that, it stayed like this forever.
The moment printer have stopped I realized the boot cannot find the file on the USB stick so logically the name of the file was wrong.I have take out the stick from the printer, put-it on my pc and I have renamed the file:
Buddy_pre_4.4_4.4.0-RC!4164.bbf
By putting the USB stick back on the printer and switching on the printer, it have booted normally,it have find the file and the update process continued normally finalizing with the update of the wifi.
This was step by step what happened.
MENTION: My MINI is not genuine but a FSYETC clone, honestly I don't know if what happened was because of that but I don't think so.
Hoping I was clear please don't hesitate if you want to know more to contact me again.
Please receive all my best wishes and thank you for contacting me.
Sincerely yours
Catalin GORETCHI - FRANCE+33.758.45.49.97
P.S. I am not French, I am from Romania but I am living in France, this is the way my name is not a French one

Prusa-Support, 2022-10-03:
Hello Catalin, thank you for the detailed report on this aspect. We'll keep checking if we might receive similar feedback from other users. I wouldn't know at this stage if the fact the printer was a clone might have changed anything, but we'll keep checking. Of course, in case of anything to further review in the meantime, do not hesitate to ask. 

Alessandro Pantaleo
Prusa Research

JazzX, 2022-10-09:
Hi,

i can also confirm this issue, on my original prusa mini.

Prusa-Support, 2022-10-12:
Hello, thank you for the additional reports! Just to make sure, were these also solved with the renaming, right? @seb123212 @JazzX 

Alessandro Pantaleo
Prusa Research

JazzX, 2022-10-12:
Yes, renaming the file fixed the problem.

seb123212, 2022-10-13:
@Prusa-Support 
Yes the problem is solved after renaming the file

TheMoen, 2022-10-30:
Weird that you two did not have the same issue as me. My prusa mini did not like any of those two filenames In 4.4.0 RC1. I removed special characters for it to work.

JazzX, 2022-10-30:
@TheMoen 
The difference may be the OS and the language of it, where you downloaded and renamed the file. I have done it on my mac with german locales.

Prusa-Support, 2022-11-01:
Hello, thank you for this update, good to hear it was resolved in all cases so far, but noted these different causes for our devs, hopefully it can be something we can improve for future, stable releases. 

Alessandro Pantaleo
Prusa Research

clawfire, 2022-11-05:
Had the same issue on my genuine mini+ . Renaming the file solved the problem 😃

SirScythe, 2022-11-25:
Hi, I just hit the same issue on the final release firmware (mini_release_boot_4.4.0+4180.bbf/mini_release_boot_update_pre_4.4_4.4.0+4180.bbf). 
Replacing the `+` by a `_` also fixed it, though it may as well could have been power cycling the printer that fixed it for all i know.

closed 2024-06-18 by github-actions[bot]