prusa3d/Prusa-Firmware-Buddy #2544: [BUG] <Footer displays wrong items>

ErGeeBee, 2022-09-30:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version** - [4.4.0-RC1+4164]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Sensor]

**USB drive or USB/Octoprint**
Printing via USB

**Describe the bug**
  When entering the new 'Footer settings' menu, the items displayed in the menu do not correspond with the actual footer items displayed in the footer.

**How to reproduce**
 Enter the Footer settings, scroll though the different items and look at the actual footer changing

**Expected behavior**
  I would expect the footer items in the menu to correspond with the actual items seen in the footer.

**Crash dump file**
  Please attach the crash dump file. This will make it easier for us to investigate the bug.

**Video**
  Please attach a video. It usually helps to solve the problem.

https://user-images.githubusercontent.com/114698077/193223630-c374008b-8ba5-4b61-9ada-6171ddd62248.mov

Gyrxiur, 2022-10-01:
Also after reboot it doesn't remember it's settings.

bmorgenthaler, 2022-10-03:
Reporting the same issue on v4.4.0-beta2 and here are how they map out on my Mini+

- Nozzle = Nozzle
- Bed = Bed
- Filament = Filament
- F Sensor = Speed
- Speed = Live-Z
- X Axis = Sheet
- Y Axis = Filament Sensor
- Z Axis = X Axis
- Z Height = Z Axis
- Print Fan = Z Height
- Hotbed Fan = Z Height
- Live-Z = Fan RPM (can't tell which)
- Sheets = Fan RPM (can't tell which)
- none = none

Eldenroot, 2022-10-23:
I can confirm - 4.4.0-RC1 - settings is also lost after reboot.

Would be really nice to have all these values with icons displayed together on one page... moreover during printing and in the main page - about?

Prusa-Support, 2022-11-26:
The issue should be fixed by the new [FW 4.4.0](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.0) stable release now. Please check it out.

Michele Moramarco
Prusa Research

closed 2022-11-26 by Prusa-Support

your-friend-alice, 2023-01-27:
still present in 4.4.1:

[image]

Eldenroot, 2023-01-27:
It is fixed. You have to set them by yourself in printer setttings...

your-friend-alice, 2023-02-01:
oh, maybe i'm misunderstanding, are there supposed to be 2 rows in the footer? I thought the second row above with the hard-coded temperature display was a bug

Prusa-Support, 2023-02-09:
It's a new feature.
You can read about _footer settings_ in the relevant release notes.
https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.0

Michele Moramarco
Prusa Research