prusa3d/Prusa-Firmware-Buddy #2545: [BUG] <Enter comprehensive title>

meermanr, 2022-09-30:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type**: MINI

**Printer firmware version**: 4.4.0-RC1+4164

**Original or Custom firmware**: Original

**Optional upgrades**: Filament runout sensor.

**USB drive or USB/Octoprint**: PrusaLink upload and start via USB

**Describe the bug**: On-printer display's "Z" number does not change during a print. In fact it shows an unchanging negative value I believe to be my Z-axis offset.

**How to reproduce**: Print something.

**Expected behavior**: The on-printer display's middle value with the "Z" in the icon would show the current Z-axis position in mm above home (i.e. the print bed surface). As the nozzle moves up and down to compensate for non-flatness detected by mesh bed levelling I would expect to see this number changing slightly, to change a bit more during z-hop travels, and to generally drift upwards as the print extends from the build plate over layers.

**G-code**: [3h_dur_transfer_custom_0.8n_0.3mm_PLA_MINI_45m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/9682432/3h_dur_transfer_custom_0.8n_0.3mm_PLA_MINI_45m.gcode.zip)

(Note this G-code has some hand edits to work around lack of M601 Pause support in Prusa MINI - specifically I set the nozzle temperature to 1ºC (below ambient room temp) so the printer waits for ever while I manhandle the print. I then use the on-printer controls to set the nozzle temperature to something more reasonable and the print resumes.

**Crash dump file**: N/A

**Images**: 

Display showing Z-axis position as -1.3. This number does not change. Also it is not the usual precision I was expectingL

[image]

Display showing my Z-axis offset, which seems to perfectly match the previous screenshot - both showing -1.3.

[image]

Firmware info: 

[image]

Also, is it intentional that the nozzle temperature is displayed twice in the bottom row?

murk-sy, 2022-09-30:
This is related to the custom footer settings - which appear to be out of order. Don't think filament sensor is supposed to be in percent

[image]

I think defaults should be **Speed or Z height (not including correction), Sheet and Filament**.
Those should be reasonable for most I believe.

krupkaj, 2022-10-03:
It seems the names in the menu does not fit to displayed values.

Prusa-Support, 2022-11-03:
Hello, thank you for the report and sorry for the late response from my side, all noted from us, we'll look into it. 🙂 

Alessandro Pantaleo
Prusa Research

Prusa-Support, 2022-11-25:
This issue will be closed as a duplicate of #2544.

Michele Moramarco
Prusa Research

closed 2022-11-25 by Prusa-Support