prusa3d/Prusa-Firmware-Buddy #2546: [ENHANCEMENT] Use clearer X/Y/Z position icons for footer

murk-sy, 2022-09-30:
**Printer type** - [MINI]
**Optional upgrades** - N/A

**Is your suggested improvement related to an existing problem? Please describe.**
Currently the icons are incorrect and can also be confusing, as the coordinate system is actually not as it's displayed by the shape
[image]

**Describe the expected functionality**
I imagine the icons would just be fixed with the correct ones later, but using one of these options would be a lot clearer at a glance, plus having the coordinate system turned correctly would be more intuitive for people unfamiliar with the coordinate system (not that they would then need the numbers?)

The alternative is also to just display the letter in orange - would be clearest for those who need it without any directional ambiguity.

The lowest row is more an experiment of how Y should be displayed, but there's limited space to work with anyway.

[image]

Eldenroot, 2022-10-23:
Really cool idea, should be changed together alongside with this #2544

Prusa-Support, 2022-11-25:
This issue will be closed as a duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2544.

Michele Moramarco
Prusa Research

closed 2022-11-25 by Prusa-Support