prusa3d/Prusa-Firmware-Buddy #2548: [BUG] gcode with non-ascii characters do not work

OniriCorpe, 2022-10-03:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MINI

**Printer firmware version** - [4.4.0-RC1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.0-RC1)

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor & Wifi

**USB drive or USB/Octoprint**
  I used an USB flash drive

**Describe the bug**
  The printer refused to open a gcode with non-ascii characters in its file name, such as this one:
`Décaèdre_0.32mm_PLA_MINI_2h15m.gcode`

**How to reproduce**
  Use the file name `Décaèdre_0.32mm_PLA_MINI_2h15m.gcode`or any other one with non-ascii characters

**Expected behavior**
  The file is used as usual

**G-code**
[Décaèdre_0.32mm_PLA_MINI_2h15m.txt](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/9694116/Decaedre_0.32mm_PLA_MINI_2h15m.txt)

**Crash dump file**
  N/A

**Video**
  N/A

Prusa-Support, 2022-11-05:
Thanks for clarifying.
Our developers will look into this.

Michele Moramarco
Prusa Research

bkerler, 2023-12-08:
Related / Double to issue #2289

closed 2024-06-18 by github-actions[bot]