prusa3d/Prusa-Firmware-Buddy #2564: [ENHANCEMENT] Continue printing until infill on filament runout sensor trigger

murk-sy, 2022-10-04:
**Printer type** - MINI+
**Optional upgrades** - Filament Runout Sensor

**Is your suggested improvement related to an existing problem? Please describe.**
Currently, when filament runout is triggered, the printing is interrupted regardless of where the print is (I believe at least?). This can cause artifacts on the surface, either due to ooze or due to nozzle not being precisely primed.

Not sure if any new version changed this, but I've had a handful of cases where filament runout ended up causing a visible mark on the surface.

**Describe the expected functionality**
After the filament sensor is triggered, a filament limit is set. If infill is reached within that limit (ideally it would be after a few infill moves, so any possible blobs are in the middle of an infill), filament change routine is triggered.

The distance between the filament sensor ball and the actual gears of the extruder is about 160mm (stock configuration!). Let's assume a safe distance of 120mm.

A 100% infill 0.20mm thick layer on the mini requires ~2700mm of filament.
A 150mm 3 perimeter cylinder with 20% grid infill requires 330mm of filament, 270 of which are the infill itself.
If filament runout is triggered just as perimeters are started, they can safely be printed to completion.

[image]

Additionally, this type of system could also help against false positives by rechecking the sensor after x mm, though I'm not sure how many - if any- there are.

**Arguments against**
- Filament end is generally bent and this could cause a jam if it gets too close to the gears (can be avoided by reducing the distance)
- Filament return may push the filament sensor off the sensor-extruder tube
- Requires user-settable runout sensor distance to prevent issues with custom setups (or to disable outright)

Prusa-Support, 2022-10-13:
Thank you for the suggestion.

I personally love the idea, but the arguments against speak for themselves.
About it, I find the listed downsides not that likely to happen, and maybe negligible or reasonable tradeoffs.
What concerns me the most is that the user will be bothered to pull off the PTFE tube every time to reach the remaining filament, because the filament will always be shorter than the distance between the extruder and the tube inlet.

Our developers will do all the necessary considerations before thinking of implementing anything like that.

Michele Mormarco
Prusa Research

murk-sy, 2022-10-13:
The PTFE tube would actually not be an issue. The PTFE tube between the extruder and the hotend is 275mm (+hotend and other stuff in between), while the filament roll to extruder PTFE tube is 150 mm (+30 mm covered by filament sensor).
Basically, you still have at least 95 mm sticking out to pull on in the *most extreme* case.

As a side note, the current "leftovers" are about 520mm long, so the extruder and the hot end add another 65 mm to the full distance.

Even someone who replaced the filament roll to sensor tube with a longer one (funnily enough, I'm one of them) would only *very* rarely need to remove it. In that sense I think it's a reasonable tradeoff, especially if the threshold can be set.

Prusa-Support, 2022-10-19:
Oh sorry. Sure, I didn't consider the extruder-to-nozzle distance which is probably 360 mm or so (275mm extr-to-head tube + about 70mm print head + 10/20 mm "padding"), more than enough for the unloading.

The "extrudable" distance though is the distance between the filament sensor triggering point and the extruder, which is slightly more than 150 mm as per the stock printer setup.
It could be enough to complete the perimeters being printed when the filament outage is detected.
Knowing and taking into account the filament sensor distance can also help reduce some filament leftovers in general, if the printer will keep extruding a bit further.

Sounds good! Our developers will evaluate the feasibility of this enhancement.

Michele Mormarco
Prusa Research

JustAnother1, 2022-10-27:
Just thinking out loud:
The printer could look for the comments in the gCode (";TYPE:Perimeter" / ;TYPE:Internal infill/...) to detect if it is currently safe to stop. It would then continue printing even if the runout sensor has fired if it is printing a perimeter.
Detecting how much plastic gets used by the additional perimeter lines should be easily detectable by loking at the "E" value in the G1 G-Code. Or by looking at the Esteps generated. 
If the perimeter is not finished after using the 15 additional centimeters of filament then a stop needs to be done anyway. This could be either due to a very long perimeter (Model with thin walls and basically no infill for example) or a slicer that forgot about the infill comment. But even in this situation the situation is better than before. We would still have a potential imperfection on the perimeter but we would have used 15 cm more of filament. So less wasted plastic!

And if the infill starts in the 150mm of plastic used then we can hide the imperfection in the middle of the Model.

murk-sy, 2023-11-30:
Regarding the Prusa XL:
Considering it's actually got 2 filament sensors and a relatively short extruder-nozzle path, I think it would be good to have an option for the printer to continue printing. With that, several options open:

The good:
- Keeping track of current filament left
- Only triggering runout before a toolchange when next layer uses more filament than available (meaning you get a prime before the next time nozzle prints, so no runout mid-layer resume related blobs/gaps)
- Informing user ahead of time "filament runout in x minutes"
- Theoretically less waste

The bad:
- Bowden tube *must* be removed at the print head to remove filament (which is clumsy and may reduce lifetime of bowden tube)

The maybe solution for bad:
- When there is only 10mm or so filament left above the extruder, just ask the user to load new filament and use the new filament to purge the ~80mm of filament 770mm^3
- However, that would take about a minute or more depending on the filament, during which the user has to be fairly vigilant as to what's happening.

murk-sy, 2024-04-10:
Since I've used spool join a few times already, I think it would be great to use it in cases like that as well.

XL has plenty of spare filament and could likely print the entire layer without issue. XL's reverse bowden tube length is 1275mm, which is enough for a 120x120x0,2mm fully infilled square. This would, however, require removing the tube at the toolhead at the end of the print. It can compromise it's long term durability of the tube's connection somewhat, but since filament is only pulled through it, in my mind it's unlikely to be much of an issue.

Additionally, if the runout toolchange is forced after a layer is completed, assuming a wipe tower is used, the new layer is guaranteed to be primed correctly, ensuring a seamless joint. Of course the other option is to force it when printing infill, which should also work perfectly fine.

Prusa-Support, 2024-04-20:
Thanks, that's surely a good point in my opinion, however, as you highlighted, we may not want to lead the user to unhook the PTFE tube too often. This should be carefully considered and thoroughly tested.
Moreover, I can't tell if forcing the interruption on infill or wipe tower is really feasible but I love the idea.

With the Prusa XL around - if this won't pose any problem to the hardware - it may be a good time to try to escalate this request.

Michele Moramarco
Prusa Research

CZDanol, 2024-04-21:
Assigned internal ticket BFW-5417. This doesn't mean that we will do it, but we will at least have a talk about it in the office :)

Have a nice day, guys!

murk-sy, 2024-04-27:
Just sort of thinking out loud, also in connection with the "unload filament after print" and "use higher temperature when purging to lower temperature filament" feature requests.

## Filament states
Filament should in my mind have the following states:
### EMPTY
There is no filament in the system

### LOOSE
There is filament detected by the filament sensor but the extruder motor is not engaged (before loading, after unloading.
Note that a broken filament sensor, detached ptfe tube due to bent filament tail (Mini) too short runout (XL) can cause filament sensor not to be triggered. This state should always be entered after unload, and theoretically set to EMPTY if filament sensor disengages afterwards.

### ENGAGED
The filament is engaged by the extruder motor, but has not reached the hotend. Cold unload can be triggered, but it cannot be extruded. This is theoretically the state immediately after filament stamping.

### PURGING
The filament is currently being purged. Ideally, this state, based on the selected material, would also select the purging temperature. For example, if you load PC, unload it, and then later load PLA, PC temperature should be used for purging. This would a variable to be stored somewhere at least.
I'm sure Prusa's R&D has done some work if this is actually required,

### IN_HOTEND
Equivalent to the usual "loaded" state, Cold extrusion is by default not 

## Filament runout
Filament runout should also have states that would likely make it easier to work with.
Having some form of filament tracking (i.e. selecting "1kg spool" or "unknown" when loading may be useful information for very long prints but that's a separate conversation.

### FILAMENT_UNLOADED
Filament is unloaded

### FILAMENT_LOADED
Filament is loaded and no runout has been triggered

### FILAMENT_RUNOUT_EXPECTED
(XL) The Advance filament sensor has been triggered, but there should still be plenty of filament. Notifying the user ahead of time with potentially a time estimate would be nice (or if the print can finish without fully running out).
Tracking remaining filament should be started from this point on.
Since Mini has a lot of filament slack, this could should be used there to reduce filament leftovers.
NOTE: How this would work with MMU is not considered since I don't have experience with it.

### FILAMENT_RUNOUT_IMMINENT
Continue printing ideally until infill or until wipe tower if possible.

### FILAMENT_RUNOUT_NOW
Mirrors the functionality of the current filament runout - after current gcode is processed, immediately start filament change procedure.

Ideally, if a print finishes with FILAMENT_RUNOUT_EXPECTED or FILAMENT_RUNOUT_IMMINENT, the filament *should* unload automatically. How to handle failed unloads is however a question I don't know how to answer. Should success be assumed and the hotend cooled down?

## Filament reserve calculation
[image]
The distances are as follows:
- HE - Hotend to extruder distance (maximum the extruder can pull back)
- ES - Extruder to extruder filament sensor distance
- So - "Standard" filament sensor to outside distance - i.e., the minimum amount of filament that has to be reversed from the moment of triggering to be able to grab it
- S1S2 - Distance between extruder and outside filament sensor (XL)
- ESo - Distance between the internal extruder sensor to outside the filament tube coupler (so you can grab it)

**Below are napkin math numbers, beware!**

### Prusa Mini
HE = 275
ES = 150
So = 30

Filament Reserve = MIN((HE - So), (ES))
Filament Reserve = MIN((275 - 30), 150)
Filament Reserve = 150 mm

Bowden tube printers especially benefit from this since the minimum reversing distance HE is so large.

### Prusa XL
HE = 75 (Prusa nozzle)
ES = 40
S1S2 = 1275
ESo = 50
So = 50

Filament Reserve (external sensor) = MIN((HE - So), (ES+S1S2))
Filament Reserve (external sensor) = MIN((75 - 50), (40+1275))
Filament Reserve (external sensor) = 25 mm

I don't think this is correct but bear with me. The math for MK4 should be pretty similar.

After side/advance filament sensor triggers:
Filament Reserve (removing PTFE at extruder) = S1S2-ESo
Filament Reserve (removing PTFE at extruder) = 1275 - 50
Filament Reserve (removing PTFE at extruder) = 1225 mm

You can print loads with that much left over.

Hopefully this can be of use or work as a baseline.

LuckyResistor, 2024-05-03:
@murk-sy Regarding the issue with removing the Bowden tube, I hope one 3D-printer manufacturer will implement the following filament switch system:

[image]

murk-sy, 2024-05-03:
Turns out even I wasn't the first, whoops

- https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1279

Considering this issue has a lot more information I'd recommend closing that one as stale

murk-sy, 2024-07-04:
Nah dawg

legomind, 2025-03-03:
Another option would be to continue printing to use up most of the reserve filament, then when it is ideal to do so, park the nozzle so it is not over any printed items and purge the filiment through the hotend, after which the user can load new filiment, and continue purge to clear the remaining remnants of the old filimament.