prusa3d/Prusa-Firmware-Buddy #2566: [BUG] Squeaking noise on Y Axis after upgrade to 4.4.0

urbiworx, 2022-10-04:
**Printer type** - MINI

**Printer firmware version** - 4.4.0

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Sensor, revo hotend, Bondtech extruder

**USB drive or USB/Octoprint**
 USB Drive

**Describe the bug**
  After upgrade I get a squeaking noise on the y axis before the print starts. It looks like the bed is being pushed a bit more than possible.

**How to reproduce**
  Print with 4.4.0 on my printer

**Expected behavior**
 No noise
**G-code**
  Reproducible with any g code I produced

**Crash dump file**
 No dump file

**Video**
  If it is unclear I will be happy to make one.

Prusa-Support, 2022-10-13:
Thank you for the report.

If the printer is using custom firmware, we may not be able to reproduce the issue and find a solution. If this is the case, please consider closing this issue and reaching the author of the modified firmware about this.

For now though, we can't exclude problems related to hardware or assembly. A video will surely be of help to see what exactly the problem looks like and possibly to understand the root cause of it.

Michele Moramarco
Prusa Research

urbiworx, 2022-10-19:
Hi,

I have used the RC1 firmware provided by prusa. No custom firmware. Here is the video: https://we.tl/t-4wTZiwMuGR

urbiworx, 2022-10-21:
[Powerful Blorr (23)_0.2mm_PLA_MINI_28m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/9841989/Powerful.Blorr.23._0.2mm_PLA_MINI_28m.gcode.zip)
Here is an example gcode producing the problem

urbiworx, 2022-10-23:
It was an assembly error. Can be closed.

Holen Sie sich Outlook für iOS<https://aka.ms/o0ukef>

Prusa-Support, 2022-10-29:
Thank you, we appreciate your feedback.

closed 2022-10-29 by Prusa-Support