prusa3d/Prusa-Firmware-Buddy #2579: [BUG] Random translation during filament change

Mryck, 2022-10-05:
**Printer type** - [MINI]

**Printer firmware version** - 4.4.0 RC 1
 
**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor]

**Describe the bug**
With the mini in French language when the filament is already in the sensor to early during a filament change it ask for "mot de passe" with means "password" instead of remove filament or something like that 

**How to reproduce**
 Put mini in French change filament and insert filament before it's ask to

**Expected behavior**
  A clear and concise description of what you expected to happen.

I can put image if needed

Prusa-Support, 2022-10-14:
Thank you for reporting.

Because 'Prusa Research' is the direct maintainer of this language, I will forward this translation mistake to the competent department.

Michele Moramarco
Prusa Research

Prusa-Support, 2024-07-19:
Please allow me to close this issue as a duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2883 and potentially solved.

Michele Moramarco
Prusa Research

closed 2024-07-19 by Prusa-Support