prusa3d/Prusa-Firmware-Buddy #2586: [BUG] Ethernet connection lost after reboot 4.4.0-RC1

kaklik, 2022-10-07:
**Printer type** -
 MINI

**Printer firmware version** 
[4.4.0-RC1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.0-RC1)

**Original or Custom firmware** 
Original

**Optional upgrades**  
Filament Runout Sensor

**USB drive or USB/Octoprint**
 
USB flash drive 

**Describe the bug**
Suddenly unable to connect to the printer after next day boot-up

**How to reproduce**
Flash new 4.4.0-RC1 firmware, do not print anything, just look at the Prusaconnect local webpage (By entering the API key) and switch off the printer for next day trial. 
After that, the browser is not able to connect to the printer's web page again. 

**Expected behavior**
  A clear and concise description of what you expected to happen.

**G-code**
 The issue is not related to any code 

**Crash dump file**
There is nothing like a crash dump

**Video**

Here is the screenshot, from PrusaSliced, which is not able to connect to. 

[image]

I checked the IP address of the printer multiple times and it is still the same. 

[image]

I also tried to eliminate the possible issue with the local network, but even a direct connection of the PC to the printer (and providing the DHCP address from PC) does not solve the problem. 
The diagnosis is significantly made difficult by ignoring pings by the printer, which is weird but seems to be normal.

murk-sy, 2022-10-10:
Is Prusalink enabled in the Prusalink submenu? If it is already enabled, how about disabling and re-enabling?

kaklik, 2022-10-11:
After reenabling the PrusaLink the Prusaslicer responds by HTTP 409 Conflict, to the "upload and print" request. Deleting the previous code from the printer and running the Upload and Print command again does not resolve that situation.  Turning off the printer and turning it on again, does.

murk-sy, 2022-10-11:
409 error was an issue in 4.4.0-BETA2, and was resolved for me in RC1, but this seems different.

Can you test if turning off prusalink off, restarting, turning it back on and then trying to print produces the same error? Make sure you're on the main screen, not on any submenus.

kaklik, 2022-10-11:
> Can you test if turning off prusalink off, restarting, turning it back on and then trying to print produces the same error? Make sure you're on the main screen, not on any submenus.

That procedure seems to reproduce the issue only in case there is some submenu opened on Printer's display. (I can not disprove if I previously had a submenu opened or not). Additionally, the conflict 409 error appears only in the case of the "Upload and Print" request. The "Upload" only command finishes correctly.

murk-sy, 2022-10-11:
That is intentional, see 4.4.0-BETA2 release notes:

> *MINI+ must be on the home screen, “print-preview” screen, “print-finished” screen - this is intentional for safety reasons

I would leave this issue open for devs/prusa support to look at, since the original issue could mean some setting isn't saved properly immediately after upgrade.

Prusa-Support, 2022-10-20:
Thank you guys for reporting the issue and for the help.

I seem to understand that the issue was either PrusaLink being toggled off, or the MINI printer not being on _home screen, “print-preview” screen, “print-finished” screen_.

Observant of the fact that the printer must be on these screens - I wonder if the "other" (main) problem persists, or if showed up again a few times after it was clarified. Namely, if PrusaLink still happens to be toggled off automatically every once in a while... I'm not able to reproduce the issue (tested via wifi and ethernet), but I may be missing something.

Michele Moramarco
Prusa Research

kaklik, 2022-10-21:
I suggest closing it as currently unreplicable.

closed 2022-10-21 by kaklik