prusa3d/Prusa-Firmware-Buddy #2612: [BUG] Out of RAM

espr14, 2022-10-13:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MINI

**Printer firmware version** - 4.3.4

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor (disabled)

**USB drive or USB/Octoprint**
USB drive

**Describe the bug**
Malloc error - out of RAM.

**How to reproduce**
I've run approx. 100 short G codes from a file and sending few G codes over USB serial between each run. No LAN connection, No WiFi module. [MGN USB flash](https://www.walmart.com/ip/USB-Flash-drive-Silver-1GB-Flash-Drive-Waterproof-Flash-drive-jump-drive-drive-for-data-storage-and-backup-suitable-for-Windows-7-8-10-MAC/714815793).

**Expected behavior**
FW should not run out of RAM.

**G-code**
```
G1 F30000

G1 Z5
G1 Z15
G1 Z5
G1 Z15
G1 Z5
G1 Z15
G1 Z5
G1 Z15
G1 Z5
G1 Z15
```

**Crash dump file**
[dump.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/9780362/dump.zip)

**Video**
Just a red screen.

Prusa-Support, 2022-10-21:
Thank you for reporting this.

Something may be taking up too much RAM memory.

Am I understanding correctly that you are alternating g-code files running via USB flash drive and g-code files running over USB serial connection via computer?

Are you able to intentionally reproduce the error?
Are you able to identify a specific pattern of actions executed before the error appears?
I may have questions to help you figure out.

.

Can you tell or estimate how long the printer has been operating since the last reboot, or simply how long it has been continuously powered ON?
Can you estimate how long the printer has been connected via USB serial?

What's the size of the last file that was run?
Could you estimate the maximum, minimum, and average size of the recently used files before the error appeared?
What is the total size of all the files currently stored on the USB stick?

Has the error disappeared after a printer reboot?
Has it disappeared after disconnecting the USB cable or the USB stick?

Would the error appear after a certain number of prints?
Would it appear in relation to a specific gcode?
Would it appear if USB-cable-connected-and-USB-stick-disconnected, or vice versa, or only if both are connected?

We got a similar issue reported quite recently https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2351, but I wouldn't close this one as a duplicate for now because the already existing one could be related to Prusa Link (not necessarily).

Michele Moramarco
Prusa Research

espr14, 2022-10-22:
> Am I understanding correctly that you are alternating g-code files running via USB flash drive and g-code files running over USB serial connection via computer?

Yes.
> Are you able to intentionally reproduce the error? Are you able to identify a specific pattern of actions executed before the error appears? I may have questions to help you figure out.

It happens once and I haven't tried to reproduce it.
> Can you tell or estimate how long the printer has been operating since the last reboot, or simply how long it has been continuously powered ON? Can you estimate how long the printer has been connected via USB serial?

It could be hour running.
> What's the size of the last file that was run? Could you estimate the maximum, minimum, and average size of the recently used files before the error appeared? What is the total size of all the files currently stored on the USB stick?

All the files were around 90 bytes. Total could be few MBs.
> Has the error disappeared after a printer reboot? Has it disappeared after disconnecting the USB cable or the USB stick?

I've rebooted and it never appear again. But I was not doing the same batch again.
> Would the error appear after a certain number of prints? Would it appear in relation to a specific gcode? Would it appear if USB-cable-connected-and-USB-stick-disconnected, or vice versa, or only if both are connected?

I don't know. I don't have time to trace the bug (and fix it).

Prusa-Support, 2022-10-29:
In the case of the issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2351, this error can be reproduced but quite erratically. In that issue, what seems to increase the chances of the error to show up is sending a set of probably overlapping requests to the printer via browser, through Prusa Link.

Maybe a number of subsequent commands sent in a short time somehow stress the printer, either way, in a similar fashion.

This issue could actually be a duplicate. We may keep both open because the error was provoked in different ways, but perhaps either issue will lead to the same conclusion.

Michele Moramarco
Prusa Research

jonmlee, 2022-11-28:
[](url)
[dump.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/10099520/dump.zip)

Got the out of memory error when trying to load up the web interface which caused my Prusa Mini to die. It's connected via ethernet.

globaltommy, 2023-08-18:
Just happened to me too :/

closed 2024-06-18 by github-actions[bot]