prusa3d/Prusa-Firmware-Buddy #2616: [BUG] <Enter comprehensive title>

vidra0, 2022-10-16:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version** - 4.4.0 RC1

**Original or Custom firmware** - Original

**USB drive or USB/Octoprint** -  USB

**Describe the bug**
 Temperature is set to 170°C not 260, but 260 is displayed on the toolbar.

**How to reproduce**
  1. Load filament
  2. Preheat -> ABS
  3. Temperature is set to 170°C and not 260. 260°C is displayed as target in the toolbar. 170°C is set as actual target (Settings -> temperature)

**Expected behavior**
 Temperature set to 260°C or toolbar updated to 170°C
[image]
[image]

murk-sy, 2022-10-16:
This is actually intended behaviour to be immediately ready for ABL when a print starts, but I do agree in a vacuum it doesn't really make sense. It says one number in the preheat menu, but it doesn't actually heat to that temperature.

It is, however, the temperature used for first layer calibration, so in a way the text is reused from that. Displaying just 170 may actually be more confusing to the user, as they might think "wait, I can't print at just 170°C, that makes no sense!". I made some suggestions to improve it in #1795, you could see if that would make more sense for you as well.

Prusa-Support, 2022-10-17:
Hello, thank you for the report and comment! I indeed confirm the info from @murk-sy, it's intended behavior, after MBL temp raises again to the material target temp. Closing this specific thread but feel free to add more to the thread indicated in the comment in case of more ways you can think for possible improvement. 

Alessandro Pantaleo
Prusa Research

closed 2022-10-17 by Prusa-Support