prusa3d/Prusa-Firmware-Buddy #2631: Z axis goes beyond the top (Prusa Mini)

backslashV, 2022-10-20:
I was moving the Z axis using the menu on the device and noticed that it tried to go beyond the top limit of the Z rod. I immediately turned off the printer as the motor was struggling. I didn't try other axes, but the firmware should disallow this IMO to prevent damage to the printer.

ThomasCZ, 2022-10-20:
I noticed the same issue. When Z axis is on the top and I select Load filament, Z axis is trying to go even higher. This is 100% reproducible.

murk-sy, 2022-10-22:
Duplicate of #1340 
Also related #414

This happens if the printer is (re)started and not homed, or z axis is moved manually during that time.
If Z is near the top on start and then filament load is started (which raises Z by a small distance), the same issue appears. Can easily happen after a failed filament load/unload which requires turning the printer off or restarting it to manually clear filament.

(At least it used to, personally I haven't tried reproducing it on newest firmware. I do believe I've seen some tests regarding crash detection being played with.)

Prusa-Support, 2022-10-24:
Hello, thank you for the report. I indeed confirm the info from @murk-sy, I'll close this particular thread as duplicate of #1340 but do feel free to add more comments on that thread in case. Thank you. 

Alessandro Pantaleo
Prusa Research

closed 2022-10-24 by Prusa-Support