prusa3d/Prusa-Firmware-Buddy #2636: [BUG] <Visual glitches in buttons - Czech lang>

Eldenroot, 2022-10-23:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI+]

**Printer firmware version** - [e.g. 4.4.0-RC1]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor]

**USB drive or USB/Octoprint**
  USB flash drive

**Describe the bug**
  Visual glitches - buttons

**How to reproduce**
  Please describe steps to reproduce the behavior.
1) WIFI setup - button is not full width, something is on the right side
2) System info - missing part of the word "Konec" (Czech), button is too short

**Expected behavior**
  A clear and concise description of what you expected to happen.
- Both should be fixed, align the text and button width

**G-code**
---

**Crash dump file**
---

**Video**
- check images
[image]
[image]

Prusa-Support, 2022-10-25:
Hello, thank you for the report, all noted and passed on to our devs for them to review and fix it! 

Alessandro Pantaleo
Prusa Research

Eldenroot, 2023-02-05:
4.4.1 still valid

closed 2024-06-18 by github-actions[bot]