prusa3d/Prusa-Firmware-Buddy #2641: Z axis movement when loading/unloading filament

rvegele, 2022-10-24:
When selecting to replace (unload) filament after a ‘tall’ print the z axis keeps on ramming (going up) even though the top has been reached. 

Choosing to load filament right after the z axis again keeps ramming upwards even though it’s already at the top.

Firmware 4.3.4

Eldenroot, 2022-10-24:
Still happens, not good, could damage the printer

Prusa-Support, 2022-10-24:
Hello, thank you for the report, we are aware of this, though we don't expect it to cause damage to the printer. This thread is a duplicate of issue #1340, where we explained in more detail the situation on this regard. 

Closing this particular thread as duplicate, but do feel free to add more comments on #1340. Thank you. 

Alessandro Pantaleo
Prusa Research

closed 2022-10-24 by Prusa-Support

Eldenroot, 2022-10-24:
OK, but still... you buy this premium printer and you have to see this ugly bug? :D Come on :)

Prusa-Support, 2022-10-25:
Thank you for the feedback! Unfortunately, I have to confirm, I took a look at past issues and noticed an even older issue on the same aspect, #699, unfortunately it is not planned to be improved due to a high amount of more higher priority bugs. 

Copy-pasting the comment from the thread #699 from our dev, "we considered this is not bug which need to be fixed for right function of printer. So this issue have low priority. It can be avoided by lowering Z height or autohome before loading/unloading filament. Even if it happen printer won´t be damaged". 

Thank you for understanding and sorry for the negative outcome on this aspect. In case of anything do not hesitate to let us know, anything can be always reviewed from our side! 

Alessandro Pantaleo
Prusa Research