prusa3d/Prusa-Firmware-Buddy #2642: [BUG] Fan is not spining

ThomasCZ, 2022-10-24:
**Printer type** - MINI

**Printer firmware version** - 4.4.0-RC1

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint**
USB flash drive

**Describe the bug**
Printer shows "Fan is not spinning" over and over again and I am not able to print anything. Unfortunately, it looks like the printer is not able to flash back 4.3.4. Could you please fix this ASAP?

**How to reproduce**
It happens very often in my case.

[dump.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/9849293/dump.zip)

[Core_0.6_0.1mm_ASA_MINI_2h3m.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/9849295/Core_0.6_0.1mm_ASA_MINI_2h3m.zip)

smstroble, 2022-10-25:
You can flash back to 4.3.4, the printer won't automatically prompt to flash when the 4.3.4 firmware is on the USB stick but if you double tap the selector knob right after power up you can trigger the flashing process. Make sure the [Buddy_4.3.4+2835.bbf](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.3.4) file is the only firmware file on the USB stick so that it is selected when flashing.

The process is documented [here](https://help.prusa3d.com/article/firmware-updating-mini-mini_124784#128715) under `Alternative method - older firmware release`.

Using this method I was successfully able to flash back to 4.3.4 from 4.4.0-RC1 without issue.

Prusa-Support, 2022-11-01:
Thank you all for reporting this and for helping.

I can only confirm what was already said until now.
This is actually a duplicate of issue #2314. The issue was apparently improved but not totally fixed. Probably the fix didn't take into account that materials like ASA may make sure of a very slow print fan speed.
Downgrading the firmware or disabling the fan check can help for now. Meantime, our developers will look into this.

Would you mind closing this issue as a duplicate? The other issue will stay open so you can share your considerations there.
We are looking at print fan settings but the issue is the same.

Michele Moramarco
Prusa Research

Eldenroot, 2022-11-01:
@Prusa-Support - I am just curious - all tickets market with "4.4.0-RC1/beta label" will be fixed in 4.4.0 final version? Why dont you use "Milestones" feature or something like that to easier track the bugs and planned features for normal users :)

Prusa-Support, 2022-11-05:
The label '4.4.0-RC1' means this problem concerns such firmware releases.
I can't promise it will be fixed in the next or the final version, possibly yes.
Managing milestones and goals is up to the developers. Some of their goals are not immediately reflected on GitHub, they have their own management tools too.
We (from Support) help users and developers, and mediate.

Michele Moramarco
Prusa Research

pilotak, 2022-12-16:
i can confirm the save behaviour - at low speed the fan error pops in and stops the print many times during printing

murk-sy, 2022-12-16:
@pilotak The issue should be resolved unless your fan is actually being significantly slowed down at an already very low setting. Are you running the latest stable version?

pilotak, 2022-12-16:
yes, i even upgraded to v4.4.1-RC1 and same thing happend again. When printing PLA there is no problem, PETG is. I have not had any problems with 4.3.x.

pilotak, 2022-12-23:
on my side it failed just after about 2mm in height

Prusa-Support, 2023-01-03:
I'd like to know the minimum speed that causes the error and the actual RPM.

Please try to reproduce the issue intentionally by setting a print fan power at various speeds between 10% (25/255) and 100% (255/255) via _settings > temperature > print fan_?
Then check the RPM value for the problematic fan speed (or slightly higher to avoid the error), visible in _info > system info_. Possibly also slowly move the print-head and gently wiggle the cables behind to see if the RPM value will vary in various positions of the X and Z axes, or of the cables themselves.

Michele Moramarco
Prusa Research

KAD370, 2023-01-10:
Min fan speed 10% shows errors during prints, but when it is set to 15%, no issues.
But there is bigger issue with using noctua fan for cooling heat sink. Since in fw is speed set too low, noctua fans need little help to start spinning, also when try to print ASA, it overheats heat sink. It would be nice to add noctua fan option in fw that runs at 100%.

murk-sy, 2023-01-10:
Noctua fan is not a stock config so I wouldn't have too high hopes for it being added in the short term considering there's a lot of other stuff on the roadmap. But there is a separate issue for that where you should post, see https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/997

KAD370, 2023-01-10:
> Noctua fan is not a stock config so I wouldn't have too high hopes for it being added in the short term considering there's a lot of other stuff on the roadmap. But there is a separate issue for that where you should post, see https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/997

Just too bad, that they didn't show any interest in 2 years old issue.

Since there are all shematics of board here on github, I'm thinking to modify  some hardware by adding external components for running 12 or 24V fans. Those stock 5V ones are low quality fans. On my printer I can hear only loud heat sink fan.

murk-sy, 2023-01-10:
Things like that generally don't get integrated slowly, they're either on the backlog or being worked on - there's been a lot of things that were fairly suddenly added like that.

As for the fan error you're having, please report actual measured RPM - see Prusa-Support reply for details.
**However, if you are using a noctua fan while getting errors, then that is expected behavior** - fan check is not (currently) made to work with non-stock fans that run at lower rpm that expected.

Prusa-Support, 2023-01-21:
The feature _altfan_ prevents the heat-sink fan from spinning 100%.
Maybe disabling it will help. In any case, we can't guarantee the correct functionality for non-Prusa genuine parts.
https://help.prusa3d.com/article/experimental-menu-mk3-s-_161213#altfan-detection

Other than this, the fan may not be spinning in the expected range.
https://help.prusa3d.com/article/err-extr-fan-error_2156
https://help.prusa3d.com/article/err-print-fan-error_2159
This could be due to dirt/worn fan or because a third-party fans compatibility issues.

Our official support can help you troubleshoot genuine Prusa parts.
https://help.prusa3d.com/article/customer-support_2287

It seems to be confirmed that this issue is a duplicate of #2314, and will therefore be closed.
Feel free to keep your questions coming but also notice that we may be looking at faulty / non-compatible fans, moreover this may not be the relevant issue for your case.

Michele Moramarco
Prusa Research

closed 2023-01-21 by Prusa-Support