prusa3d/Prusa-Firmware-Buddy #2649: [BUG] PrusaLink ERR_CONNECTION_REFUSED

ThomasCZ, 2022-10-24:
**Printer type** - MINI

**Printer firmware version** - 4.4.0-RC1

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor

**Describe the bug**
  I am unable to open PrusaLink with WiFi any more. There is always `ERR_CONNECTION_REFUSED` message in Chrome or Edge browser.

Eldenroot, 2022-10-24:
I can confirm, yesterday all fine, today not working.

Eldenroot, 2022-10-24:
Check the flash drive, is it fine?

ThomasCZ, 2022-10-24:
Yes, it is fine, at the moment, the printer is printing from USB drive. Looks like, it is independent from WiFi connection. I uploaded `gcode` to printer when no USB drive was inserted (https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2627) and there was no error message.