prusa3d/Prusa-Firmware-Buddy #2656: Z-axis home fails, says PINDA issue after selftest.[BUG] <Enter comprehensive title>

dhissink, 2022-10-27:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Prusa Mini+

**Printer firmware version** 4.4.0 RC1

**Optional upgrades** -  Filament Runout Sensor

**USB drive or USB
 

**Describe the bug**
 After upgrading firmware to 4.4.0RC1 The selftest fails, telling Z-home not correct. possible PINDA problem or not connected. However, bed leveling goes fine.  The message also has a QR -code but leading to an 'Error 404' page.

**How to reproduce**
  Start selftest

**Expected behavior**
 No failure message

**G-code**
 No G-code other then selftest

**Crash dump file**
  ?

Prusa-Support, 2022-10-31:
Hello, thank you for your message. Can you please send me a video of the full self-test, from start to fail, and a picture of the error message QR code, that ended up in the 404 error? Either here, or directly to me, info@prusa3d.com mentioning me. It might indeed be a bug on the 4.4.0 RC1 specifically, but with this info we can take a look in more detail and try to resolve it before the final 4.4.0 release. Thank you. 

Alessandro Pantaleo
Prusa Research

dhissink, 2022-10-31:
> Hello, thank you for your message. Can you please send me a video of the full self-test, from start to fail, and a picture of the error message QR code, that ended up in the 404 error? Either here, or directly to me, ***@***.*** ***@***.***> mentioning me. It might indeed be a bug on the 4.4.0 RC1 specifically, but with this info we can take a look in more detail and try to resolve it before the final 4.4.0 release. Thank you.

Hello Alessandro,

Thank you very much for your fast reply. The bug happened only once and I cannot reproduce it anymore.  It seems the printer is functioning as it should now.
Many thanks,
Dick

closed 2024-06-18 by github-actions[bot]