prusa3d/Prusa-Firmware-Buddy #2663: [ENHANCEMENT] Actual time when printing

ThomasCZ, 2022-10-27:
The printer is showing time of complete instead of remaining time when is connected to WiFi. I think, there is also very important to know the actual time. Could you please show it somewhere (maybe like an option in footer?) or add option to choose remaining time and time of complete?

Thanks

Eldenroot, 2022-10-28:
[image]

espr14, 2023-12-18:
### Like in [Snake FW](https://github.com/Snake-FW/P32-FW/releases)?

[image]

Prusa-Support, 2024-01-17:
This issue will be closed as a duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/648 and https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1322.

Michele Moramarco
Prusa Research

closed 2024-01-17 by Prusa-Support