prusa3d/Prusa-Firmware-Buddy #2665: [BUG] Missing translations (Czech) - print screen

Eldenroot, 2022-10-28:
**Printer type** - [MINI+]

**Printer firmware version** - [4.4.0RC1]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor]

**USB drive or USB/Octoprint**
  print via USB flash drive

**Describe the bug**
  A clear and concise description of what the bug is.

**How to reproduce**
  Please describe steps to reproduce the behavior.

**Expected behavior**
  Screen during the print (starting) - there are displaying texts which are not translated - for example heating bed, e heating, e cooling, bed cooling etc.

**G-code**
  ---

**Crash dump file**
  ---

**Video**
  ---

Eldenroot, 2022-10-28:
I will add some photos, give me few minutes.

Eldenroot, 2022-10-28:
[image]
[image]

[image]
[image]
3 photos - you can see English text, last photo - just to see that I use Czech language

Kalimero13, 2022-10-28:
Tahle chyba je tam bohužel už několik let.

Eldenroot, 2022-10-30:
> Tahle chyba je tam bohužel už několik let.

Thats why I had created the report :) this should be fixed. Czech product but missing translation :)

Prusa-Support, 2022-10-31:
Hello, thank you for reporting this, I passed this on to our devs, hopefully it can be fixed soon. 🙂 

Alessandro Pantaleo
Prusa Research

closed 2024-08-12 by github-actions[bot]

CZDanol, 2025-01-15:
Hey, we are slowly migrating to a new print status message API, so the messages should get eventually all translated - this will possibly happen over a few releases.