prusa3d/Prusa-Firmware-Buddy #2677: [BUG] <Missing translations (Czech) - Lišta/footer settings>

Eldenroot, 2022-11-02:
**Printer type** - [MINI]

**Printer firmware version** - [4.4.0_RC1]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor + Bondtech IFS extruder]

**USB drive or USB/Octoprint**
  USB

**Describe the bug**
  Some items are not translated - for example the word "ITEM"

**How to reproduce**
  Open Lišta (Czech) or I think it is "Footer" in English setting page

**Expected behavior**
  "Item" word should be translated, check the image below.

**G-code**
  ---

**Crash dump file**
  ---

**Video**  
[image]

**Crash dump file**
  ---

**Video**
  ---

Eldenroot, 2022-11-04:
This is weird - translation exists for these "Item" but it is not used

https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/master/src/lang/po/cs/Prusa-Firmware-Buddy_cs.po#L591

Prusa-Support, 2022-11-10:
Thanks for the report.
Something may have gone wrong during the translation operations - I can't tell.
I will let the competent team know.

Michele Moramarco
Prusa Research

Prusa-Support, 2024-07-26:
It seems that similar translation inconsistencies have been totally overridden when the MINI's firmware was merged with the newer MK4/XL version so this issue may have potentially auto-fixed.
A confirmation from Czech firmware users would be appreciated.

Michele Moramarco
Prusa Research

Prusa-Support, 2024-08-17:
Closing as potentially solved now, but please comment below if you think this issue is still valid.

Michele Moramarco
Prusa Research

closed 2024-08-17 by Prusa-Support