prusa3d/Prusa-Firmware-Buddy #2678: [BUG] Missing translations for experimental menu (Czech)

Eldenroot, 2022-11-02:
**Printer type** - [MINI]

**Printer firmware version** - [4.4.0_RC1]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor + Bondtech IFS extruder]

**USB drive or USB/Octoprint**
  USB

**Describe the bug**
  Experimental menu is not translated (Czech).

**How to reproduce**
  Open experimental menu.

**Expected behavior**
  I think that "Experimental menu" should be translated. Also the direction phrase "wrong" should be renamed to "reversed".

**G-code**
  ---

**Crash dump file**
  ---

**Video**
[image]

Eldenroot, 2022-11-02:
Also this notice is not translated - it is displayed when you wanna save the new values in the menu (Experimental settings)

[image]

Prusa-Support, 2022-11-24:
Hello, thank you for reporting these missing translations, I passed this on internally so that it can be corrected asap. 

Alessandro Pantaleo
Prusa Research

closed 2024-09-30 by github-actions[bot]