prusa3d/Prusa-Firmware-Buddy #2679: [BUG] <Satin sheet profile is missing>

Eldenroot, 2022-11-02:
**Printer type** - [MINI]

**Printer firmware version** - [4.4.0_RC1]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor + Bondtech IFS extruder]

**USB drive or USB/Octoprint**
  USB

**Describe the bug**
  I updated to 4.4.0_RC1 (from 3.3.1 I think some time ago). I didnt have the "Satin1" and "Satin2" sheet profiles displayed. So I read this article from your release note:

[image]

However this is not true, I performed factory reset and I still have Custom1, Custom2, Custom3, Custom4 but no Satin1 or Satin2!

**How to reproduce**
  IDK, I have 4.4.0_RC1 flashed and I performed factory reset.

**Expected behavior**
  Custom 3 and 4 should be changed to Satin1 and Satin2?!

**G-code**
  ---

**Crash dump file**
  ---

**Video**
[image]

Prusa-Support, 2022-11-03:
Hello! Thank you for the report, passed it on to devs. 🙂 

Alessandro Pantaleo
Prusa Research

Eldenroot, 2022-11-24:
Is there anybody else with same issue? I updated directly from 4.3.1 to 4.4.0_RC, maybe this is the issue?

krupkaj, 2022-11-26:
Just installed 4.4.0 and there are still 4 customs sheets, not a satin.

Eldenroot, 2022-11-26:
Yes, exactly...

Eldenroot, 2022-11-26:
And no feedback but they promote this feature in the changelog

Prusa-Support, 2022-11-26:
Hello, I am sorry for not having more feedback on this regard so far. I passed this on to our devs, they'll check on it as soon as possible to correct the situation on this aspect. 

Alessandro Pantaleo
Prusa Research

Prusa-Support, 2022-12-09:
Hello! Thank you for letting us know, reported to devs, hopefully it can be fully fixed in the future release. 

Alessandro Pantaleo
Prusa Research

Eldenroot, 2023-01-15:
Any progress on this? I have many plates (I use for them custom 1-2-3 profiles)... would be nice to get satin name in profile :)

mivanov, 2023-01-17:
I see the same issue after upgrading directly to version 4.4.1 (from some 2021 version I think)

JanTuts, 2023-02-03:
I also have the same issue after upgrading firmware from 4.3.4 to 4.4.1, and then performing a factory reset.
I still had the "Satin" profiles right up to doing that factory reset, but afterwards they were "Custom".

Prusa-Support, 2023-02-03:
Hello! Thank you for making us notice, I reproduced this as well on 4.4.1, and reported this discrepancy to devs! 

Alessandro Pantaleo
Prusa Research

Eldenroot, 2023-02-05:
Good, finally! :)

Eldenroot, 2023-06-22:
Prusa Mini is sacrificed by PRUSA devs... they only support Mk4 and 3... so next  time buy different printer.

mivanov, 2023-06-22:
I gotta mostly disagree on that, even if this is a bit annoying it's quite a minor issue to need to remember that "Custom"="Satin". On balance I still love the Mini, it is rock solid and not sure there are comparable options at this quality and footprint. Not that I have needed to look since it still works perfectly. I still trust Prusa will continue to support the Mini with updates more than I would trust another company (my low expectations for this particular issue notwithstanding 😄 )

Prusa-Support, 2023-09-09:
The _Satin_ profiles were introduced with FW 4.3.4 but were accidentally removed in recent tims. This should change again in the near future.

As a temporary fix, you can downgrade to [4.3.4](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.3.4) (perform a factory rest only if the Satin profiles won't appear) and then upgrade again to most recent [4.4.1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.1), however if you perform a factory reset the _Satin_ profile will be lost.

These profiles are pre-set and can't be renamed but please notice, except for the naming, there are no differences between them. The user may save any value on any profile, regardless of what the sheet profile is called.

Michele Moramarco
Prusa Research

CZDanol, 2024-05-16:
Closing in favor of #1318