prusa3d/Prusa-Firmware-Buddy #2680: [BUG] Wizard - the temperature for bed/nozzle is wrong during heating phase test

Eldenroot, 2022-11-02:
**Printer type** - [MINI+]

**Printer firmware version** - [4.4.0_RC1]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor + Bondtech IFS extruder]

**USB drive or USB/Octoprint**
  USB

**Describe the bug**
  During self test in wizard the nozzle and bed are heated - the temperature is rising but the value after the "/" is wrong - it displays XX/"30". Check the attached image.

**How to reproduce**
  Turn on wizard with self test after the factory reset.

**Expected behavior**
  It should display the temperature which is set for heating.

**G-code**
  ---

**Crash dump file**
  ---

**Video**
  [image]

Prusa-Support, 2022-11-24:
Hello, thank you for the report, passed this on to our devs to better review and fix it! 

Alessandro Pantaleo
Prusa Research

Eldenroot, 2023-02-05:
This is same if you run the test from the menu (not from the wizard).

You cannot see the set point for nozzle temperature - it says XX/0 - always "0" displayed. It is not good, very confusing

closed 2024-06-20 by github-actions[bot]