prusa3d/Prusa-Firmware-Buddy #2681: [ENHANCEMENT] Experimental menu - add a reverse option for the filament sensor (Bondtech IFS extruder)

Eldenroot, 2022-11-02:
Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - [MINI+]
**Optional upgrades** - [Filament Runout Sensor + Bondtech IFS extruder]

**Is your suggested improvement related to an existing problem? Please describe.**
  When you use the new version of Bondtech IFS Extruder for Prusa Mini/Mini + (the older version of Bondtech extruder is discontinued), you have to switch the wires in the connector (filament sensor) - switch red and black wire. 

**Describe the expected functionality**
  I believe this can be solved in firmware. This makes sense to me to add this new setting to the experimental menu.

Prusa-Support, 2022-11-09:
Thanks for your suggestion.
We are talking about compatibility issues with third-party components but our developers may consider such requests in the case ofo popular mods.

Michele Moramarco
Prusa Research

rinkek, 2022-12-18:
To add to popularity... I would like this option too !!

Please....

closed 2024-06-20 by github-actions[bot]