prusa3d/Prusa-Firmware-Buddy #2682: [BUG] Image from g-code file is not rendered on printer screen

Eldenroot, 2022-11-02:
**Printer type** - [MINI+]

**Printer firmware version** - [4.4.0_RC1]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor + Bondtech IFS extruder]

**USB drive or USB/Octoprint**
  USB

**Describe the bug**
  Image for g-code file is not displayed, I see only text and requirements for filament

**How to reproduce**
  Check the attached file.

**Expected behavior**
  Image should be displayed.

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.
[tension-meter_petg_mini.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/9924546/tension-meter_petg_mini.zip)

**Crash dump file**
  ---

**Video**
  Please attach a video. It usually helps to solve the problem.

[image]

derenma, 2022-11-10:
@Eldenroot I just looked at your g-code and the thumbnails aren't in there. (They would be base64 encoded blobs at the top of the gcode)

This is likely a slicer issue, me thinks.

Eldenroot, 2022-11-10:
Maybe, if yes close this one :)

closed 2022-11-24 by Eldenroot