prusa3d/Prusa-Firmware-Buddy #2690: [BUG] Print end time - 12/24H format issue

Eldenroot, 2022-11-05:
**Printer type** - [MINI+]

**Printer firmware version** - [e.g. 4.4.0_RC1]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [Filament Runout Sensor + Bondtech IFS extruder + Wifi module]

**USB drive or USB/Octoprint**
  USB

**Describe the bug**
  Weird funny issue - we use in our country 24H format for time. When you start printing in the morning but the print itself will continue to the afternoon -> the time is displayed in 12H format. For example:

A) start the print 9:56 (morning, for US it is 9:56AM), total time for printing is 6 hours
B) end time should be 15:56 (afternoon, for US it is 3:56PM), BUT:
C) the end time is displayed as 3:56, this is not correct! 
As I described this now means that the print will last not 6 hours BUT 18 hours!

**How to reproduce**
  Check above.

**Expected behavior**
  It should follow 12/24H format and do not change automatically to make it confusing. 

**G-code**
  ---

**Crash dump file**
  ---

**Video**
  I started the print at 9:56 (morning), it takes 6 hours... there should be displayed end time 15:56, NOT 3:56! 
[image]

Eldenroot, 2022-11-05:
OK, here is the new info, it is even more interesting and funny!

- somehow it is now correct, after some time (few minutes) it just changed from 3:56 to 15:56 (15:57). It is same print, photo took few minutes after the one from the first post in this ticket. Maybe it did some recalculation (it is one minute longer) and this fixed also the 12/24H time format issue?!

[image]

Maybe it is related how the time from the internet is synchronized?

derenma, 2022-11-10:
@Eldenroot I was just about to post this same bug however, I still read 03:56 as 24H time, not 12H time. Meh. Not a big observation on my part, but _in theory_ my interpretation could point to another block of code for the devs to check.

Eldenroot, 2022-11-10:
Yes, but it should finish at 15:56 :)

derenma, 2022-11-18:
@Eldenroot I am curious: Are you seeing issues like I am here: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2732 ?

I am wildly speculating that these issues are connected.

Prusa-Support, 2022-11-24:
Hello! Thank you for the post, reported this to our devs! 

Alessandro Pantaleo
Prusa Research

Eldenroot, 2022-11-24:
@Prusa-Support - thank you, anyway I like this feature BUT:

- you should display both times - relative (18 minutes remaining) and absolute 18:56 print finish time
Also you should add a clock to the header on the display.

closed 2024-06-20 by github-actions[bot]