prusa3d/Prusa-Firmware-Buddy #2691: [BUG] Printer freeze during the first layer calibration (check/insert the filament screen)

Eldenroot, 2022-11-05:
**Printer type** - [MINI+]

**Printer firmware version** - [e.g. 4.4.0_RC1]

**Original or Custom firmware** - [Original]

**Optional upgrades** - [e.g. Filament Runout Sensor + Bondtech IFS extruder + WIFI module]

**USB drive or USB/Octoprint**
  USB

**Describe the bug**
  During the first layer calibration the printer freeze and you cannot do anything. Only way is to use the reset button. It happens only sometimes, not on every first layer calibration. It is related somehow to the screen where the info about filament - insert/continue instructions are displayed.

**How to reproduce**
  First layer calibration (wizard or in the settings where you can set the first layer for every bed type - custom1/2, satin etc.).

**Expected behavior**
  Printer should not freeze.

**G-code**
  ---

**Crash dump file**
  ---

**Video**
  --

Eldenroot, 2022-11-24:
Still happens, only me?

Prusa-Support, 2022-11-25:
Thanks for reporting but I'm sorry, I'm unable to reproduce it.

Maybe, during this specific process, some error occurred resulting in an interruption instead of an error displaying.
If I recall correctly I've seen that in the past: anomalies caused by errors unable to be displayed during some calibration process. I'm not even sure what was the printer model but an anomaly like this wouldn't be a surprise.

A video may help.

No errors if you launch a self-test?
Would any error show up during a normal print?

Does the issue still occur if the filament sensor is physically disconnected?
Does it occur if the wifi module is physically disconnected?

Did you notice any thermal anomaly?
Any error if you preheat the printer, slide nozzle and bed, and gently wiggle their cables every few centimeters?

Does the issue occur in the same way if you downgrade to 4.4.0-beta or 4.3.4?
Would other kinds of anomalies manifest during the same process?

Michele Moramarco
Prusa Research

sander1095, 2023-01-28:
Hello!

I just received and assembled my printer and went through the wizard. I can report I have the same issue as well.

I have a mini+ and updated to 4.4.1.

After resetting the printer, things worked fine!

Eldenroot, 2023-02-05:
-Tested with 4.4.1 firmware, still valid!

Prusa-Support, 2023-02-09:
Please, videos and more background information could help understand and reproduce the problem.
.
@Eldenroot I wonder if your other issue https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2842 is somehow related.

Michele Moramarco
Prusa Research

Eldenroot, 2023-02-09:
Nope, this is different. This happens only when you somehow put the filament inside during the first layer calibration and you use filament sensor.

Prusa-Support, 2023-02-16:
Oh ok. If so, I'm afraid the instructions for the reproducibility of the problem are insufficient.
A video may help. Seeing, not only the LCD screen, but also all actions performed by the user, could be of help.

Michele Moramarco
Prusa Research

closed 2024-06-20 by github-actions[bot]