prusa3d/Prusa-Firmware-Buddy #2700: [BUG] Double extruder dial on lcd screen after flashing 4.4.0 RC

Oxize, 2022-11-08:
**Printer type** - [MINI]

**Printer firmware version** - [4.4.0 RC]

**Original or Custom firmware** - [Custom]

**Optional upgrades** - [Filament Runout Sensor]

**USB drive or USB/Octoprint**
  USB/Octoprint

**Describe the bug**
  After flashing firmware 4.4.0 RC you see an double temp extruder dial on the lcd screen.

**How to reproduce**
  Flash firmware 4.4.0 RC

**Expected behavior**
  Should be only 1 temp dial for the extruder.

**G-code**
  not needed

**Crash dump file**
 Not needed.

[image]

Eldenroot, 2022-11-10:
Duplication of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2544 and https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2545

Prusa-Support, 2022-11-25:
Thanks for reaching us.
Unfortunately, most of the footers deserve a correction.

This issue will be closed as a duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2544.

Michele Moramarco
Prusa Research

closed 2022-11-25 by Prusa-Support

Eldenroot, 2022-11-25:
Will be there another RC release or we are waiting for 4.4.0 final? Thx

your-friend-alice, 2023-01-27:
still present in 4.4.1:
[image]