prusa3d/Prusa-Firmware-Buddy #2705: [BUG] Filament change procedure with filament sensor is needlessly rigid

MightyPork, 2022-11-08:
**Printer type**
MINI+

**Printer firmware version**
4.4.0-RC1

**Original or Custom firmware**
Original

**Optional upgrades**
Filament Sensor

**USB drive or USB/Octoprint**
USB drive

**Describe the bug**

When replacing filament, when you remove the old one and insert new one before clicking OK on the control panel ("Was unloading successful?"), it then forces you to remove the new filament again and insert it second time.

**How to reproduce**

1. Ask to replace filament, use a filament change GCODE, or have filament run out mid-print to trigger the replacement procedure
2. See that the old filament is out, pull it out of the PTFE tube and filament sensor
3. Insert new filament into the tube through the filament sensor to be loaded
4. Press OK on the control panel

**Expected behavior**
I think there is no need for the filament change procedure to check the filament sensor at all. Alternatively, it can detect that the filament was already replaced. It can automatically close the prompt and proceed to ask if new filament is inserted.

murk-sy, 2022-11-09:
I rarely change filament (except on runout) so I haven't really noticed this to be an issue for me, but detecting and caching if filament was removed while the window is active would probably be the best solution.

Another option is to automatically close the "is unload successful?" prompt if filament sensor detects the filament was removed. There are some edge cases where that might not be optimal (filament breaking between extruder and hotend, or malfunctioning sensor), but it would likely be a QoL improvement for users with your filament change procedure.

Prusa-Support, 2022-11-29:
Thanks for the suggestions.
The current behavior is intentional and not a bug, but there is surely room to improve.
Indeed it would make sense for the "unloading successful" question to automatically disappear if the filament sensor gets untriggered after such a question.

Michele Moramarco
Prusa Research

closed 2024-06-19 by github-actions[bot]