prusa3d/Prusa-Firmware-Buddy #2706: [BUG] Missing CZ translation of text "Probing Point" when bed mesh leveling.

pacaj2am, 2022-11-08:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MINI

**Printer firmware version** - 4.4.0-RC1

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint**
n/a

**Describe the bug**
Missing CZ translation of text "Probing Point" when bed mesh leveling.

**How to reproduce**
Start print with CZ language

**Expected behavior**
CZ translation should be shown, "Měřím bod" in consistency with MKš (https://github.com/prusa3d/Prusa-Firmware/blob/e109da4db9148fc5a36a971815322f69a2393954/lang/po/Firmware_cs.po#L2032)

**G-code**
n/a

**Crash dump file**
n/a

**Picture**
[image]

Eldenroot, 2022-11-10:
Duplication of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2665

Prusa-Support, 2022-11-29:
This issue will be closed as a duplicate.

Michele Moramarco
Prusa Research

closed 2022-11-29 by Prusa-Support