prusa3d/Prusa-Firmware-Buddy #2718: [ENHANCEMENT] Display actual TIME on print screen + improve the behaviour when you are connected to internet

Eldenroot, 2022-11-13:
Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - [MINI+]
**Optional upgrades** - [Filament Runout Sensor + Bondtech IFS extruder + WIFI module]

**Is your suggested improvement related to an existing problem? Please describe.**
  With latest 4.4.0_RC1 firmware is nice to have an option to use WIFI and see the finish time of printing... but I cannot see the actual time, so it is quite frustrating to check every time another clock. Moreover would be really cool to extend this feature and display remaining time together with finish time! Now you lose one of these if you are connected to internet.

**Describe the expected functionality**
  Display actual time in the header, next to the Wifi/Lan icon. Display both - time remaining + finish time together on print screem.

mix579, 2022-11-13:
Of these idea, my preference is for time remaining. The actual time is nice but as long as I know how  much time is left on the print I'm okay. Plus, actual time would have to deal with daylight savings time, and unless it reads time from the internet, which it doesn't seem to do, it would be more confusing than helpful. 

Anyway, I agree with the OP, at this point it's hard to see how much time is left on the print.

Prusa-Support, 2022-12-08:
There is an already active discussion about the time behavior - strictly related to the internet connection, starting with FW [4.1.0-RC1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.1.0-RC1) (_Support for the NTP servers_) - here https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/648.

Also about the request of displaying the current time, here https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1322

The topic is a bit redundant as you may notice from issues like https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1067, https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1287 and a few more recent relates ones.

I suggest making use of already existing issues to discuss this problem/request, in order for neater management of the issues and higher efficiency in the development planning.
For the same reason, consider posting every single problem/request in separate issues: one issue could be sorted while the other may remain open. In fact, improving the UI and improving the time behavior could be separate tasks in the development workflow.

I hope you don't mind me closing this issue as duplicate/redundant.
If you guys think this issue should necessarily stay open separately, please let me know.

Michele Moramarco
Prusa Research

closed 2022-12-08 by Prusa-Support