prusa3d/Prusa-Firmware-Buddy #2719: [BUG] Wrong time and progress

ThomasCZ, 2022-11-14:
**Printer type** - [MINI]

**Printer firmware version** - 4.4.0-RC1

**Original or Custom firmware** - Original

**Describe the bug**
Printer did calibration and waiting for reach temperature but time and progress was wrong. I expected zero progress and estimated time. Printer showed that the part will be finished after 5 weeks.

[image]

Gordontgopher, 2022-11-15:
Is it possible that the progress just represented the start code (set temps, bed levelling etc.) within the file that had been completed?

ThomasCZ, 2022-11-15:
No, it looks like, the displayed progress is counting during heating and leveling and it is reset when printer actually starts printing. For example, printing takes 10 minute and heating 1 minute. When heating is completed, the printer shows 10% progress.

reedy, 2022-11-22:
Similar to #1067 (amusingly filed similar time of year, with a similar completion date)

closed 2024-06-19 by github-actions[bot]