prusa3d/Prusa-Firmware-Buddy #2720: [BUG] Printing will be finished in past

ThomasCZ, 2022-11-14:
**Printer type** - MINI

**Printer firmware version** - 4.4.0-RC1

**Original or Custom firmware** - Original

**Describe the bug**
Photo was taken in 22:43. The printer has internet connectivity, I expect relative accurate time estimation.

[image]

ThomasCZ, 2022-11-14:
Time zone is set correctly

mix579, 2022-11-14:
Mine was also set correctly but when we switched over from Daylight Savings to Standard time, my printer didn't update accordingly, and so my times are all one hour off. Until I change the time zone manually. Looks like unlike my phone the Prusa doesn't switch between DST and regular time automagically.

neophyl, 2022-11-15:
Time syncs are usually done to an underlying UTC time (which is what it grabs from the internet servers) and then your machine applies a local offset of that UTC time based on what you configure on the device.  
Its quite common for embedded designs to NOT have any compensation for daylight savings.  
As to implement this requires the device to have a database of countries that do and do not use daylight savings.  So not only do you have to set your time zone but also a city location too usually.  
The databases themselves would be fairly large (for an embedded device) and a complicating factor is that the contents of the database have to be updated regularly as you wouldn't believe how many places change if they are using daylight savings from year to year.  
Basically its a pain in the posterior to develop and maintain an auto daylight saving compensating device.  My car doesn't even do it :)

ThomasCZ, 2022-11-15:
It looks like, time is shifted by 3 minutes. Maybe, there is some bug in time compensation of heating?

Eldenroot, 2022-11-21:
The heating time should not be included in the printing time... because it is really difficult to track all according the prusaslicer where you can see the time axis and compare the printing easily with the model.

reedy, 2022-11-22:
I've filed a few bugs about time issues and some about clock drift - #1321

closed 2024-06-19 by github-actions[bot]